atcollab/at #340: File save/load not possible

swhite2401, 2021-12-21:
Several issues experienced with saving and loading files:

-Sometimes the first element (ringparam) is initialized with FamName='', this prevents the lattice from loading
properly. This is consecutive to a save_mat()
-When the Particle attribute of this first element is not a particle object and does not have the proper attributes
loading also fails

I add @simoneliuzzo in this thread because he found the problem generating a lattice from scratch (I think?), he can provide more information if useful

lfarv, 2021-12-21:
- Is it specific to .mat files ?
- May I have an example of wrong file, with the way it has been generated (from Matlab of python ?). For instance, why the Particle attribute of a RingParam would not be a particle ?

swhite2401, 2021-12-21:
```
import at

q = at.Quadrupole('QUAD', 0)
ring = at.Lattice([q],energy=1,periodicity=1)

at.save_lattice(ring, './quad.mat')
ring = at.load_lattice('quad.mat')
```

This code makes it fails, if I force FamName of RingParam element to a value it seems to work (for the first issue)

lfarv, 2021-12-23:
Fixed in #341

PR #341 (2021-12-28): Fix problems reading .mat files
This fixes #340.
- Workaround a bug (?) in scipy.io.loadmat where empty strings from .mat files disappear,
- Fix problems with particle objects in .mat, .m and .repr files
closed 2021-12-28 by lfarv