atcollab/at #345: loss of TruncatedGaussian

Wanggl180328, 2022-01-08:
When I fellow the aterrcor.pdf, adding errors for elements by atsetrandomerrors, there is a mistake to find the TruncatedGaussian in matlab. And I find a function named TruncatedGaussian in github there is still error ocurrs. 
Meanwile, I would like to know if I want to add error angle in x y dirrction by my own code, how can I change the field T1 and T2.

simoneliuzzo, 2022-01-12:
Dear @Wanggl180328, 

I think you are using functions from the pubtools folder. 
No warranty is given for these functions and as you may see the development has stopped long ago. It may be resumed in the month to come, but the code is going to be completely different.
In order to use the most recent version of the atmat/pubtools/LatticeTuningFunctions, please switch to the corresponding branch errors_correction.
In this branch the function that you are looking for is available.
hope this help

best regards
Simone Liuzzo

Wanggl180328, 2022-01-12:
Dear Simone Liuzzo,

Thanks for your help. I found the function I need following your guide. And I find the pubtools useful and I am inspired to know a new version is coming soon.

Best wishes
Wang

closed 2022-01-17 by Wanggl180328