atcollab/at #353: Installing accelerator_toolbox

elafmusa, 2022-01-19:
Hello everyone,

I'm move from using Jupyter notebook to PyCharm.
I need to install the accelerator toolbox there, i used the normal PyCharm packing installation to look for the AT,
When i install it i got the error massage (below) , i tried also to use the command " pip install --user accelerator-toolbox " in the PyCharm terminal but i got the error "pip install --user accelerator-toolbox SyntaxError: invalid syntax". Could you plese help me installing the AT.

**" Collecting accelerator-toolbox
  Using cached accelerator-toolbox-0.2.1.tar.gz (151 kB)
  Preparing metadata (setup.py): started
  Preparing metadata (setup.py): finished with status 'done'
Requirement already satisfied: numpy>=1.16.6 in c:\users\musa\appdata\local\programs\python\python310\lib\site-packages (from accelerator-toolbox) (1.22.1)
Requirement already satisfied: scipy>=0.16 in c:\users\musa\appdata\local\programs\python\python310\lib\site-packages (from accelerator-toolbox) (1.7.3)
Using legacy 'setup.py install' for accelerator-toolbox, since package 'wheel' is not installed.
Installing collected packages: accelerator-toolbox
    Running setup.py install for accelerator-toolbox: started
    Running setup.py install for accelerator-toolbox: finished with status 'error'

    ERROR: Command errored out with exit status 1:
     command: 'C:\Users\musa\AppData\Local\Programs\Python\Python310\python.exe' -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\musa\\AppData\\Local\\Temp\\pip-install-7iujpsb4\\accelerator-toolbox_6e48c669af4b4b3aa04266b2db4b23a8\\setup.py'"'"'; __file__='"'"'C:\\Users\\musa\\AppData\\Local\\Temp\\pip-install-7iujpsb4\\accelerator-toolbox_6e48c669af4b4b3aa04266b2db4b23a8\\setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\musa\AppData\Local\Temp\pip-record-dzdgdou0\install-record.txt' --single-version-externally-managed --user --prefix= --compile --install-headers 'C:\Users\musa\AppData\Roaming\Python\Python310\Include\accelerator-toolbox'
         cwd: C:\Users\musa\AppData\Local\Temp\pip-install-7iujpsb4\accelerator-toolbox_6e48c669af4b4b3aa04266b2db4b23a8\
    Complete output (57 lines):
    running install
    running build
    running build_py
    creating build
    creating build\lib.win-amd64-3.10
    creating build\lib.win-amd64-3.10\at
    copying at\__init__.py -> build\lib.win-amd64-3.10\at
    creating build\lib.win-amd64-3.10\at\integrators
    copying at\integrators\pyDriftPass.py -> build\lib.win-amd64-3.10\at\integrators
    copying at\integrators\pyIdentityPass.py -> build\lib.win-amd64-3.10\at\integrators
    copying at\integrators\__init__.py -> build\lib.win-amd64-3.10\at\integrators
    creating build\lib.win-amd64-3.10\at\lattice
    copying at\lattice\cavity_access.py -> build\lib.win-amd64-3.10\at\lattice
    copying at\lattice\elements.py -> build\lib.win-amd64-3.10\at\lattice
    copying at\lattice\lattice_object.py -> build\lib.win-amd64-3.10\at\lattice
    copying at\lattice\options.py -> build\lib.win-amd64-3.10\at\lattice
    copying at\lattice\utils.py -> build\lib.win-amd64-3.10\at\lattice
    copying at\lattice\__init__.py -> build\lib.win-amd64-3.10\at\lattice
    creating build\lib.win-amd64-3.10\at\load
    copying at\load\allfiles.py -> build\lib.win-amd64-3.10\at\load
    copying at\load\elegant.py -> build\lib.win-amd64-3.10\at\load
    copying at\load\matfile.py -> build\lib.win-amd64-3.10\at\load
    copying at\load\reprfile.py -> build\lib.win-amd64-3.10\at\load
    copying at\load\tracy.py -> build\lib.win-amd64-3.10\at\load
    copying at\load\utils.py -> build\lib.win-amd64-3.10\at\load
    copying at\load\__init__.py -> build\lib.win-amd64-3.10\at\load
    creating build\lib.win-amd64-3.10\at\matching
    copying at\matching\globalfit.py -> build\lib.win-amd64-3.10\at\matching
    copying at\matching\matching.py -> build\lib.win-amd64-3.10\at\matching
    copying at\matching\__init__.py -> build\lib.win-amd64-3.10\at\matching
    creating build\lib.win-amd64-3.10\at\physics
    copying at\physics\amat.py -> build\lib.win-amd64-3.10\at\physics
    copying at\physics\energy_loss.py -> build\lib.win-amd64-3.10\at\physics
    copying at\physics\fastring.py -> build\lib.win-amd64-3.10\at\physics
    copying at\physics\harmonic_analysis.py -> build\lib.win-amd64-3.10\at\physics
    copying at\physics\linear.py -> build\lib.win-amd64-3.10\at\physics
    copying at\physics\matrix.py -> build\lib.win-amd64-3.10\at\physics
    copying at\physics\nonlinear.py -> build\lib.win-amd64-3.10\at\physics
    copying at\physics\orbit.py -> build\lib.win-amd64-3.10\at\physics
    copying at\physics\radiation.py -> build\lib.win-amd64-3.10\at\physics
    copying at\physics\ring_parameters.py -> build\lib.win-amd64-3.10\at\physics
    copying at\physics\__init__.py -> build\lib.win-amd64-3.10\at\physics
    creating build\lib.win-amd64-3.10\at\plot
    copying at\plot\generic.py -> build\lib.win-amd64-3.10\at\plot
    copying at\plot\specific.py -> build\lib.win-amd64-3.10\at\plot
    copying at\plot\synopt.py -> build\lib.win-amd64-3.10\at\plot
    copying at\plot\__init__.py -> build\lib.win-amd64-3.10\at\plot
    creating build\lib.win-amd64-3.10\at\tracking
    copying at\tracking\particles.py -> build\lib.win-amd64-3.10\at\tracking
    copying at\tracking\patpass.py -> build\lib.win-amd64-3.10\at\tracking
    copying at\tracking\track.py -> build\lib.win-amd64-3.10\at\tracking
    copying at\tracking\__init__.py -> build\lib.win-amd64-3.10\at\tracking
    warning: build_py: byte-compiling is disabled, skipping.
    
    running build_ext
    building 'at.tracking.atpass' extension
    error: Microsoft Visual C++ 14.0 or greater is required. Get it with "Microsoft C++ Build Tools": https://visualstudio.microsoft.com/visual-cpp-build-tools/
    ----------------------------------------
ERROR: Command errored out with exit status 1: 'C:\Users\musa\AppData\Local\Programs\Python\Python310\python.exe' -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\musa\\AppData\\Local\\Temp\\pip-install-7iujpsb4\\accelerator-toolbox_6e48c669af4b4b3aa04266b2db4b23a8\\setup.py'"'"'; __file__='"'"'C:\\Users\\musa\\AppData\\Local\\Temp\\pip-install-7iujpsb4\\accelerator-toolbox_6e48c669af4b4b3aa04266b2db4b23a8\\setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\musa\AppData\Local\Temp\pip-record-dzdgdou0\install-record.txt' --single-version-externally-managed --user --prefix= --compile --install-headers 'C:\Users\musa\AppData\Roaming\Python\Python310\Include\accelerator-toolbox' Check the logs for full command output**

swhite2401, 2022-01-21:
Hello

I am not a Windows user but this seems to be the problem:

error: Microsoft Visual C++ 14.0 or greater is required. Get it with "Microsoft C++ Build Tools": https://visualstudio.microsoft.com/visual-cpp-build-tools/

lfarv, 2022-01-21:
AT is distributed in 2 formats: a "source" distribution (python and C files, which have to be compiled upon installation), and "pre-compiled", so-called binary,  distribution (*.whl files).  These have been prepared for most platforms. When you call `pip install --user accelerator-toolbox`, it looks 1st for a binary version, if one corresponds to your system, and if not, for the source distribution. I am surprised that in your example it did not take the pre-compiled one, since a Windows distribution exists. May be your system is older than the one used on GitHub to compile. For confirmation, could you try `pip install --user --only-binary :all: accelerator-toolbox`, which restricts to binary distribution ? Anyway, if it appears that no pre-compiled version fits your machine, you will indeed have to use the source distribution, which requires you to install the Visual C++ compile as mentioned in the error message.

closed 2022-11-18 by swhite2401