atcollab/at #370: Impossible to import lattice with custom elements in pyAT

swhite2401, 2022-02-16:
Presently, when the class cannot be defined (from attributes or passmethod) it defaults to `at.Marker`
However, when the lattice is "sanitized", `at.Marker `accepts only `'IdentityPass'`.

This makes it impossible to import lattices with custom passmethod for which a class cannot not be defined.

A temporary fix, to avoid users to be blocked, is proposed in the branch `custom_elements` by remove the check on at.Marker which now accepts any passmethod.
Alternatively we may use at.Element as the default when the class cannot be defined and the `passmethod != 'IdentityPass'`

lfarv, 2022-02-17:
@swhite2401: Did you try the `check=False` keyword  in `load_mat` or `load_lattice`? It should solve your problem.

swhite2401, 2022-02-17:
Maybe, but there is still a logic issue, I think it makes much more sense that when a class is cannot be defined it defaults to `Element `instead of `Marker` that has a constraint on the pass method

swhite2401, 2022-02-17:
In fact I do not really understand why we should be so strict...regarding elements like marker or monitors, why can't we allow passmethods other than 'IdentityPass'? Is there a good reason for that?

lfarv, 2022-02-17:
No, I agree with you. `load_mat` was one of the 1st functions of PyAT (of course, you need data!) , and Will was very cautious in checking everything. There are only very few checks I'd like to keep:
- a `Marker` or `Monitor` should have length 0 (it's common sense)
- `IdentityPass` applies only to elements with length 0

Also note that these constraints are checked at element creation, but not if one tries to modify attributes…
Apart from that, one could relax a lot.

The `check=False `is workaround for immediate use, so that we could keep #371 at low priority (it seems that there is a lot more going on)

swhite2401, 2022-02-17:
Ok so how about we merge this one as temporary fix for our ALS colleagues to work, it anyway is an improvement and they are happy with it.

Then we keep the issue open to remind us a full clean-up is necessary

PR #371 (2022-02-18): Custom elements
This branch fixes #370
closed 2022-02-18 by swhite2401