atcollab/at #374: atdynap question

lnadolski, 2022-02-20:
Dear All,

Why the function dynaper as this split between positive and negative DA

for i=1:np+3
   [xx(i),zz(i)]=bscan(ring,nt,clorb,...
	  xpmax*cos(t1(i))*slist,zmax*sin(t1(i))*slist);
end
for i=np+4:2*np+3
   [xx(i),zz(i)]=bscan(ring,nt,clorb,...
	  xmmax*cos(t1(i))*slist,zmax*sin(t1(i))*slist);
end

instead of 
for i=1:np+2
   [xx(i),zz(i)]=bscan(ring,nt,clorb,...
	  xpmax*cos(t1(i))*slist,zmax*sin(t1(i))*slist);
end
for i=np+3:2*np+3
   [xx(i),zz(i)]=bscan(ring,nt,clorb,...
	  xmmax*cos(t1(i))*slist,zmax*sin(t1(i))*slist);
end

lfarv, 2022-02-21:
Maybe @simoneliuzzo can answer the question?

simoneliuzzo, 2022-03-21:
I am not an author of atdynap. 
I actually agree with @lnadolski . I never noticed! I do not use much atdynap, a part for quick tests.
Below two figures. 

```
v = load('~/beamdyn/EBSlattices/AT/S28F.mat');

NT = 100;

diary('out_atdynap.m')
tic;
[x,y]=atdynap(v.r,NT);
t=toc;
disp(t);
diary off;

diary('out_atdynap_indexing.m');
tic;
[xi,yi]=atdynap_indexing(v.r,NT);
t1=toc;
disp(t1);
diary off;

%% 
figure;
plot(x,y,'x-','DisplayName','atdynap','MarkerSize',12); hold on;
plot(xi,yi,'o:','DisplayName','indexing change','MarkerSize',12);
xlabel( 'x [m]' );
ylabel( 'y [m]' );
legend('Location','south');

```
[image]
[image]

closed 2025-07-23 by swhite2401