atcollab/at #380: Off-energy lattice with atx

linahv, 2022-02-28:
Hi all, 
Working on an off-energy lattice, I extensively use atx. For now, atx computes the linear optics, tunes and chromaticities with atlinopt, meaning they are not linked to the RF-related parameters unless you input the right energy deviation. Maybe it could use atlinopt6 instead ?

lfarv, 2022-02-28:
Hi @linahv: `atx` is the next function planned for upgrade with `atlinopt6`, together with `fastring`. For now, to use it off-momentum, you should be careful:
- tune the RF frequency for the desired off-momentum value: `atx` internally turns radiation on, so it needs the right frequency. `atsetcavity` does that easily in the last AT versions,
- **turn radiation off**, otherwise atlinopt gives garbage,
- specify the desired off-momentum in the 2nd input argument.

linahv, 2022-02-28:
Hi Laurent, 

Thanks, I wasn't aware of the planned upgrade. I'll follow your using steps for now !

closed 2022-11-18 by swhite2401