atcollab/at #388: atbaseplot(THERING,@plCorrectorStrength) does not works

lnadolski, 2022-03-20:
Dear @lfarv 

Another incompatibility with the new plotting format.

`atbaseplot(THERING,@plCorrectorStrength);
Error using plCorrectorStrength
Too many output arguments.
Error in xplot (line 47)
[s,outp]=plotfun(splitring,curve.dpp,plotargs{:},dpargs{:});
Error in wrapper6d (line 23)
        [varargout{1:nargout}]=func(ring,is6d,varargs{:});
Error in atbaseplot (line 57)
[varargout{1:nargout}]=wrapper6d(ring,@xplot,ax,args{:}); 
`
Laurent.

lnadolski, 2022-03-20:
I am not sure about the correction

Hearafter a proposal

function varargout=plCorrectorStrength(varargin)
%function varargout=plCorrectorStrength(lindata,ring, dpp, varargin) %#ok<INUSD>
%plCorrectorStrength    Plot corrector strengths
%
%Plots polynomB for ring and ring1
%
% See also atplot atbaseplot

if nargout == 1 % From atplot
    [lindata,ring]=deal(varargin{1:2});
    CoD=cat(2,lindata.ClosedOrbit);
    PolyBVal=zeros(size(CoD(1,:)));
    PolyBVal1=zeros(size(CoD(1,:)));

    PolynomBVal1=zeros(size(CoD(1,:)));
    PolynomBVal2=zeros(size(CoD(1,:)));
    PolynomBVal3=zeros(size(CoD(1,:)));
    PolynomBVal4=zeros(size(CoD(1,:)));
    ind=findcells(ring,'PolynomB');

    PolynomBVal1(ind)=getcellstruct(ring,'PolynomB',ind,1,1);
    PolynomBVal2(ind)=getcellstruct(ring,'PolynomA',ind,1,1);
    PolynomBVal3(ind)=getcellstruct(ring,'PolynomA',ind,1,2);
    PolynomBVal4(ind)=getcellstruct(ring,'PolynomB',ind,1,2);

    plotdata(1).values=[PolynomBVal1' PolynomBVal2' ...];%...
        PolynomBVal3'/1000 0*PolynomBVal4' ];
    plotdata(1).labels={'H cor','V cor',...
        'Skew Quad/1000','Quad'};
    plotdata(1).axislabel='PolynomB';
    % dispersion=cat(2,lindata.Dispersion)';
    % plotdata(2).values=dispersion(:,3);
    % plotdata(2).labels={'\eta_y'};
    % plotdata(2).axislabel='vertical dispersion [m]';
    varargout = {plotdata};
else % from atbaseplot
    [ring,dpp]=deal(varargin{1:2});
    [linargs,varargs]=linoptions(varargin(3:end));
    [~,lindata]=atlinopt6(ring,1:length(ring)+1,linargs{:});
    s=cat(1,lindata.SPos);
    varargout={s,plCorrectorStrength(lindata,ring,dpp, varargs{:})};
end
end

lfarv, 2022-03-24:
Fixed in #392.

@lnadolski : Thanks for your checks. Your correction should work. But since the linear optics data are not used, calling `atlinopt6` when called from `atbaseplot` is useless, so I skipped it (small benefit of `atbaseplot` compared to `atplot`).

PR #392 (2022-03-24): Fix plCorrectorStrength
Fix #388. `plCorrectorStrength` fails when called from `atbaseplot`.

This plot helper displays `PolynomB(1)`, `PolynomA(1)`, `PolynomA(2)` and `PolynomB(2)`.

**Warning**: for some reason, PolynomB(2) was scaled by 0 !!! It is now scaled by 0.001, as PolynomA(1).
closed 2022-03-24 by lfarv