atcollab/at #391: Harmonic number not initialized

swhite2401, 2022-03-23:
@liuzzo has found that in some cases, even though an RF system is defined, the `harmonic _number` properties of the `Lattice` is not initialized preventing some functionalities of AT to work properly.
The error returned is that the ring as no cavity which is obviously wrong.

In case `frequency` or `harmonic_number` are not provided to the `Lattice` constructor would it be possible to get the `frequency` from the cavity element, and only if this element is not found return an error?

lfarv, 2022-03-24:
Matlab or python ?
Can you give an example showing the problem?

swhite2401, 2022-03-26:
@lfarv, this is python, here is a piece of code to make fail:

```
import at
ring = at.load_lattice('../lattice/hmba.mat')
ring = list(ring[1:])
ring = at.Lattice(ring, energy=6.0e9)
print(ring.harmonic_number)
```

lfarv, 2022-03-27:
@swhite2401 : in the particular case, there is indeed no cavity. So this behaviour is normal. Such a lattice is perfectly suitable for all 4D optics computations, and this is checked by the test sequence by using the `dba.mat` test file which has no cavity.
On the other hand, in the 1<sup>st</sup> comment, you refer to a lattice with a cavity : this is a problem.

When thinking about it, the error message may be misleading in some cases: if the lattice is initialised without cavity (no harmonic number attribute), and if you manually insert one, there is no way to generate the harmonic number, you have to set it explicitly also. In that case, the message is wrong. One could instead make it more general: "harmonic_number is undefined".

lfarv, 2022-03-27:
The rule to derive harmonic_number is the following (theoretically…):

- if found in a `RingParam` element, take it (only when reading `.m` or `.mat` files)
- otherwise if it is defined in the lowest frequency cavity, take it,
- otherwise use the lowest cavity frequency and ring circumference,
- otherwise it's undefined.

This happens at lattice initialisation. Later on, it has to be defined explicitly.

swhite2401, 2022-03-28:
Why do you say this ring has no cavity?

```
ring = at.load_lattice('../lattice/hmba.mat')
cavs = ring.get_elements(at.RFCavity)
print(cavs)
```

Gives:
`[RFCavity('RFC', 0.0, 187500.0, 352372212.4670127, 31, 6000000000.0, PassMethod='IdentityPass')]`

swhite2401, 2022-03-28:
Ah sorry, matlab/python confusion... in fact I remove the cavity

swhite2401, 2022-03-28:
@simoneliuzzo had a failing example I cannot reproduce

swhite2401, 2022-03-28:
> @swhite2401 : in the particular case, there is indeed no cavity. So this behaviour is normal. Such a lattice is perfectly suitable for all 4D optics computations, and this is checked by the test sequence by using the `dba.mat` test file which has no cavity. On the other hand, in the 1st comment, you refer to a lattice with a cavity : this is a problem.
> 
> When thinking about it, the error message may be misleading in some cases: if the lattice is initialised without cavity (no harmonic number attribute), and if you manually insert one, there is no way to generate the harmonic number, you have to set it explicitly also. In that case, the message is wrong. One could instead make it more general: "harmonic_number is undefined".

Why not computing it using either of these 2:
- otherwise if it is defined in the lowest frequency cavity, take it,
- otherwise use the lowest cavity frequency and ring circumference

in `get_harmonic_number()` if undefined?

swhite2401, 2022-04-06:
So here is a simplified version of the case that fails. The usage of `Lattice()` is disputable (I anticipate comments) but this should not fail in any case:

```
nring = at.Lattice([quad], energy=6e9, periodicity=1)
nring.insert(len(nring),cav)  
nring = at.Lattice(nring, energy=6e9, periodicity=1)
try:
   print(nring.harmonic_number)
except AttributeError as err:
   print(err)
nring.harmonic_number = cav.HarmNumber
print(nring.harmonic_number)
```
Output:
```
harmonic_number undefined: No cavity found in the lattice
31
```
it is clear that the second call of `Lattice()` does not initialize `harmonic_number` despite the presence of a cavity. The solution being to explicitly initialize `ring.harmonic_number`

lfarv, 2022-04-06:
Got it.

- solution 1: the `insert` (and `extend`) methods scan their argument to update the Lattice attributes,
- solution 2: the `Lattice` constructor scans the lattice even if it's already a Lattice object. At the moment, it just copies the existing attributes.

Solution 1 looks better, I'll look at that.

swhite2401, 2022-04-07:
I prefer solution 1 as well

lfarv, 2022-04-29:
fixed in #401

closed 2022-04-29 by lfarv