atcollab/at #402: lattice element selection by index still acts on all FamName occurences

bi-veglia, 2022-04-28:
When trying to modify the KickAngle of a corrector selecting it by lattice index, the modification is applied to every element with the same FamName, ignoring the index. 
eg. the command 

>> ring[specific_corr_index].KickAngle = [Hangle, Vangle]

acts on the KickAngle of every element sharing the family name of the element at ring[specific_corr_index].

swhite2401, 2022-04-28:
I am not sure this is your problem but in python when you write:
```
a = some_object
ring = [a, a, a, a]
```

`ring` will be populated with pointers to `a`, in case you want independent elements you have to make sure to add copies of `a` instead.

bi-veglia, 2022-04-28:
I see. I'll think of some other ways around this, thanks

closed 2022-04-28 by bi-veglia