atcollab/at #406: PyAT linopt one-turn phase advance

T-Nicholls, 2022-05-20:
Hi all, it's been a while.

In the `lindata0` array output from linopt `mu` is always returned as 0.0, why is this?
Taking a quick look at the code it seems to be intentional as it is manually set to 0.0, this seems a little strange to me.

lfarv, 2022-05-25:
Hi @T-Nicholls, nice to hear from you! 
You're right, the phase advance is set to 0, to be consistent with the definition of `lindata0`: values et the entrance of the lattice.
And I agree that it does not bring a lot of information. One could change the definition to "values at the exit of the lattice". All other values are identical. I guess that nobody makes any use of the 0 phase advance, so this change should not disturb anybody.  This has to be modified in all versions (`linopt2`, `linopt4`, `linopt6`, `linopt`).

lfarv, 2022-06-06:
> Yes, I noticed that `linopt` has split/multiplied since I last saw it. I would be happy to update all applicable instances if, of course, people are in favour of the change in definition. Would the Matlab codebase also need changing?

This split occurred when implementing a powerful new analysis method (`linopt6`). `linopt2` and `linopt4` keep the existing method inherited from `linopt`, but with a better interface identical to `linopt6`. And of course, linopt stays unchanged, but is not recommended. The Matlab code was changed simultaneously (`atlinopt2`, `atlinopt4`, `atlinopt6`).

swhite2401, 2022-06-08:
Hello both, I am a bit skeptical about a change of definition. I am not convinced what you propose holds for a transfer line, in fact I very much prefer that lindata0 remains the values at the beginning of the line.

T-Nicholls, 2022-06-08:
Hello @swhite2401, I am not very familiar with the simulation of transfer lines so I'll have to do a little research. However, if this change would have a negative effect on the handling of transfer lines then I agree the definition should remain the same. The benefit is only a very minor convenience and so is definitely not worth impacting the behaviour of anything else in PyAT.

swhite2401, 2022-06-10:
Hi @T-Nicholls, I do not have a strong opinion but my feeling is that if you do:

```
l0, _, ld = ring.lattice_pass( )
_, _, ldl = ring.lattice_pass(twiss_in=l0)
```

`ld `and and `ldl` should be identical which is presently the case.

On the other hand if you start mixing in `l0` parameters at the end and the beginning of the lattice  you are introducing potential source of confusion.
These are essentially the same for a ring, except for the phase, but are not necessarily the same for a transfer line that is btw used as well for lattice matching.

I would therefore personally stick to the simple definition: parameters a the beginning of the lattice even though I fully agree that `mu=0.0` in this case has no added value

swhite2401, 2022-06-10:
Alternatively, you could add `l0.tune` that has a clear definition, the `beamdata` output was added recently to hold these global outputs and seemed to make sense at the time, but it can be discussed

T-Nicholls, 2022-06-28:
After taking another look, I agree that this change is not worth impacting the simulation of transfer lines for. The current definition serves a specific purpose in the case of transfer lines and my proposed definition change is only a minor convenience for ring users. So I am happy for the definition of `lindata0` to remain the same.

closed 2022-06-28 by T-Nicholls