atcollab/at #427: wrong rbend entrance and exit angles from matlab m file

oscarxblanco, 2022-06-29:
Dear at developers,

when loading a matlab .m file using the at.load_m function, there is a wrong assignment to the EntranceAngle and ExitAngle for atrbend elements, setting them to zero.

This error is not present when loading the lattice from a matlab .mat file because the function find_class in pyat/at/load/utils.py copies both attribute values.

[image]

swhite2401, 2022-06-30:
Hello @oscarxblanco , interesting find! We will take a look. @lfarv did you see this?

lfarv, 2022-06-30:
@oscarxblanco : thanks for reporting. I'm looking at that…

lfarv, 2022-06-30:
Corrected in #430

closed 2022-07-01 by lfarv