atcollab/at #441: Different calculated emittance between atx and the radiation integral  in rectangular bend

Wanggl180328, 2022-07-14:
When using atsummary and atx, the calculated emittance is highly consistant when there is only sector bends. But when there are rectangular bends, the emittance is quite different, which can be as large as 10%.
I found that the function atx using ohmienvelope to find the R matrix, and the ohmienvelope using findmpoleraddiffmatrix to calculated the radiation diffusion matrix.  But I find this function calculates a same result between the bends with and without fringing-field, which may be not correct.
In addition, the transform matrix calculated from the findm66 is different between the bends with and without fringing-field.
[image]

lfarv, 2022-07-15:
Hello @Wanggl180328.
As you noticed, `atsummary` and `atx` use 2 different methods to compute equilibrium emittances.

- `atsummary` uses the classical radiation integrals I<sub>1</sub>.. I<sub>5</sub>. None of these integrals involve the pole face angle of bending magnets. Only the bending radius and field index matter. This makes sense since synchrotron radiation occurs along curved trajectories, not at magnet faces.
- `ohmienvelope` computes diffusion matrices along the ring. Similarly, the diffusion matrix of a bending magnet depends on the radius of curvature, not on face angles. So I understand the fact that both diffusion matrices are identical.

On the other hand, the pole face angle has a strong effect on the transfer matrix, by adding focusing/defocusing at magnet faces. That's what you also noticed.

So nothing in your observations looks abnormal, except the fact that the emittance differs. Could you post an example of a full stable lattice using rectangular dipoles, showing this 10% difference ?

Side remark: in your 1st example, the bending magnet is far from rectangular: the exit angle is zero (sector), while the entrance angle is 5 times the bending angle. It should be half the bending angle, so it looks wrong by a factor 10.

Wanggl180328, 2022-07-15:
Dear lfarv,
All of my question begins at the observation of the difference of the calculated emittance. And I found in other lattice, actually, the demos in the AT code the calculated emittance is roughly the same. And the lattice I used is quite a simple one, maybe only there are  fringing fields. I am not very familiar with the Ohmi method, so I supposed the fringing fields lead to wrong calculation in this method.
I used the operating lattice in the HLS-II storage ring, it is a DBA lattice. But the calculated emittance is 37.98 nmrad using atsummary but 43.4014 nmrad using atx, which is more than 10% different. The lattice is shown below.
Your answer helps a lot for the understanding of the diffusion matrix. And I appreciate for that
[HLSlattice.zip](https://github.com/atcollab/at/files/9119410/HLSlattice.zip)
.