atcollab/at #454: pyAT: at.Mutlipole element creation issue

swhite2401, 2022-08-02:
For `at.Multipole` in case `K` is provided as input, `__init__()` fails because of a conflict with the property, this happens when saving a `.mat` lattice and then reloading it.
I suppose the property is defined here such that it is more convenient to create DQs, but this is a source of problems.
Adding the line `kwargs.pop('K', None)` in the `__init()__` of `at.Multipole` solves the problems but is not satisfactory...

Less critical: I do not understand why K is a required attribute for `at.Dipole`, can't it be made optional? I have also noticed that in the help k is defined as the 'field index', I am not sure what this means.

lfarv, 2022-08-02:
>  I do not understand why K is a required attribute for at.Dipole

`k` is an optional keyword argument. If you omit it, it is set to zero (no field gradient). But you are right concerning the help: it's the focusing strength, not the field index.

swhite2401, 2022-08-03:
> `k` is an optional keyword argument. If you omit it, it is set to zero (no field gradient). But you are right concerning the help: it's the focusing strength, not the field index.

Yes sorry it appears in the required attributes list...I got side tracked by this

lfarv, 2022-08-04:
@swhite2401 : Could you give the exact sequence that you tried? 

May be you tried to instantiate a `Multipole` element? `Multipole` in intended to be a base class, not to be instantiated. The rule is that you should use the class corresponding to the lower multipole present in the magnet:
- a DQ is a `Dipole` with a quadrupole component (crucial, otherwise you don't get the 'Bnd*' passmethod),
- a quadrupole-sextupole is a `Quadrupole` with a sextupole component
- ...

It's true that defining a pure dodecapole is ambiguous. I would use the Quadrupole class…

I'll look at what happens when instantiating a Multipole. In case of problems, there are 2 possibilities:
1. Correct the problem, if possible. At first glance, taking the `K` property out of `Multipole`, into both `Dipole` and `Quadrupole` looks simple,
2. turn `Multipole` into an abstract class

swhite2401, 2022-08-04:
@lfarv, you are right I instantiate a `Multipole`, that is in fact the problem.  This action has to be allowed, for example at ESRF lattices are all define with multipoles.
Multipole can be any arbitratry magnet and this is very convenient. For instance how would you define correctors doing H/V dipole + skew (ESRF), or corrector coils on octupoles (PSI)?

Then as you mentioned there are other high order magnets used in some facility that do not have a class, it is better to have a generic `Multipole` for these than using `Quadrupole` that will make it complicated to parse magnet types.

swhite2401, 2022-08-04:
I am very much in favor of option 1.

swhite2401, 2022-08-04:
And to answer your question the exact sequence that fails:

[image]

lfarv, 2022-08-04:
@swhite2401 : can you test #455 ? It should solve the problem, but I'm not in good conditions to test.

lfarv, 2022-08-15:
Fixed in #455

closed 2022-08-15 by lfarv