atcollab/at #463: confusing output of atx, ringpara, atsummary

simoneliuzzo, 2022-08-18:
[image]

Computing the horizontal natural emittance of a lattice with `atx`, `ringpara` or `atsummary`, shifting the rf frequency with `atSetRingProperties` the values of emittance off-energy are largely different. 

For my experience, the values of `atx` are those reproduced in Control Room with beam.

lfarv, 2022-08-18:
@simoneliuzzo: you do not indicate if your lattice has radiation ON or OFF. I suspect ON, otherwise changing the frequency makes no sense.

 At the moment, `atsummary` and `ringpara` **cannot be used with an active cavity**: the need "radiation OFF". Changing the RF frequency has no effect (`ringpara`) or unpredictable results (`atsummary`). So:

1. the figure is irrelevant,
2. the problem is indeed a real one: both `atsummary` and `ringpara` have to be updated to accept:
   - 6d lattices by using `atlinopt6` instead of `atlinopt`
   - `dp` or `dct` specification on 6d lattice by acting of the RF frequency, using the `wrapper_6d` wrapper.

This is on the todo list.
However note that even when it's done, using a 6d lattice for `atsummary` or `ringpara` has no interest: they are based on linear optics, so you'll get identical results with more computation time.

Both functions are very similar, but each has its users. So we have to keep both, but ensure that they give identical results. The comparison with `atx` is different, because its method is completely different: Ohmi's envelope tracking based formalism versus analytical radiation integrals.

lfarv, 2022-10-30:
This issue was pointing out 2 different problems:
1. `atsummary` and `ringpara` could not be used with 6d lattices, did not accept `df` input and gave different results,
2. the equilibrium emittance returned by `atx` and `atsummary` are different.

The problem 1 is solved in #501: `atx`, `atsummary` and `ringpara` accept off-momentum specification with either `dp`, `dct` or `df`. They also accepts a 6d lattice as input, though it makes no difference in the results.

With this, here is an updated figure:

[image]

This gives the answer to the problem 2: `atsummary` and `ringpara` are based on the radiation integrals in the dipoles. They cannot take into account the radiation in quadrupoles, which is responsible for most of the emittance variation when running off-momentum (anti-bend effect). So their result is very similar to the result of `ohmienvelope` if it considers only the dipoles.

**Conclusion: `atsummary` and `ringpara` cannot compute the equilibrium emittance for off-momentum lattices.**

With this, I think the issue can be closed.

simoneliuzzo, 2022-10-31:
> Conclusion: `atsummary` and `ringpara` cannot compute the equilibrium emittance for off-momentum lattices.

This comment and a mention to use ohmienvelope could be added to atsummary and ringpara help to guide the users.

simoneliuzzo, 2022-10-31:
If the results of atsummary and ringpara off-energy are false, why dpp/dct/df input can be provided? 
May be it would be more userfriendly if the emittance is computed by ohmienvelope automatically in this case?

lfarv, 2022-10-31:
Hi @simoneliuzzo . Nice to see you are not too far away…

> If the results of and ringpara off-energy are false, why dpp/dct/df input can be provided?

As far as I can see, everything is correct in the results, including the dipole radiation integrals (see the plot above), **except** the equilibrium emittance. So I propose, for off-momentum lattices, to set the emittance to NaN and print a message indicating that the equilibrium emittance cannot be computed because it misses the quadrupole contribution.

> May be it would be more userfriendly if the emittance is computed by ohmienvelope automatically in this case?

I think we should not mix things: `atsummary` is based on analytic computation, `ohmienvelope` on Ohmi's formalism, it should stay such. Each method has its pros and cons, let's give the choice to the user and give him enough information!

Does this fits you? I'll prepare a pull request based on that.

swhite2401, 2022-10-31:
Hello both, can't we improve the analytical calculation assuming the ideal dispersive orbit in quadrupoles?
This does not seem impossible no?

lfarv, 2022-10-31:
@swhite2401 : I agree, it looks possible to compute the quadrupole contribution with some approximation. But is it worth the effort ? 
Anyway, we can proceed step by step: I'll make a proposal to secure the use of at summary and ringpara, and then anybody is welcome to add the quadrupole contribution to the integrals.

swhite2401, 2023-04-06:
This was fixed in #511 and #583

closed 2023-04-06 by swhite2401