atcollab/at #464: findorbit6 failed in Ri guess not provided when radiation is off

lnadolski, 2022-08-18:
Dear @lfarv 

https://github.com/atcollab/at/blob/40c29fe1339cee8e30e6144e6a2d41802a05d60a/atmat/atphysics/Orbit/xorbit_6.m#L25

findorbit6(ring) will not work if radiation if off.
atgetU0 requires radiation ON. This call is useless since Ri per default should be zeros(6,1);

I suggest this modification

if isempty(Ri) && check_radiation(RING)
    Vcell=sum(atgetfieldvalues(cavities,'Voltage'));
    U0cell=atgetU0(RING,'periods',1,'method',method);
    if U0cell > Vcell
        error('AT:MissingVoltage','Missing RF voltage: unstable ring');
    end
    Ri = zeros(6,1);
    Ri(6) = -L0/(2*pi*HarmNumber) * asin(U0cell/Vcell)+cavities{1}.TimeLag;
else % no radiation
    Ri = zeros(6,1);
end

Best regards,

Laurent.

lfarv, 2022-08-19:
Hi @lnadolski;

I do not understand you problem: You can't get a stable longitudinal motion without a cavity. As soon as you activate a cavity, the badly named "`check_radiation`" will be True, whether there is radiation or not. And `findorbit6` works normally: `atGetU0` returns 0 if there is no radiation, and the resulting orbit is zeros(6,1).

- `findorbit6` does not (and cannot) work with "radiation" OFF: you need a cavity (but you don't need synchrotron radiation !).
- the problem lies in the bad naming of things, which is corrected in the last python pull request (`enable_6d`, `disable_6s`, `is_6d` instead of `radiation_on`,...), but not yet in Matlab. And `findorbit6` definitely needs "6d" enabled.

lnadolski, 2022-08-22:
Dear @lfarv,

In my mind `check_radiation` was wrongly associated only to radiation.
This is misleading indeed.

This issue becomes irrelevant.
I close it.

closed 2022-08-22 by lnadolski

lfarv, 2022-08-22:
You are right, the names are confusing. That's why I propose to put aliases with more meaningful names:
- `is_6d` property <=> `radiation` property (done in #443)
- `atenable_6d` <=> `atradon` (upcoming pull request)
- `atdisable_6d` <=> `atradoff`
- `check_6d` <=> `check_radiation`

And similarly in PyAT