atcollab/at #467: Protect master

swhite2401, 2022-08-25:
Hello @lfarv, git is offering protection for the master (and has been telling me that it does for a while now!), in particular request for a PR and that all test pass to merge changes. Maybe we should activate these? What do you think?

lfarv, 2022-08-25:
I agree with a protection against direct commits into "master" and forcing the use a pull request. On the other hand, I think we should keep the possibility to ignore test results and reviews for the moment. Test failures still happen…

swhite2401, 2022-08-25:
Ok done

closed 2022-08-25 by swhite2401

lfarv, 2022-08-25:
@swhite: can you tell me where you modified the project settings? Thanks

swhite2401, 2022-08-25:
Sure, see screenshot:

[image]