atcollab/at #468: at.get_value_refpts fails if the field value is scalar and an index is provided

simoneliuzzo, 2022-08-27:
`get_value_refpts(ring, array_of_indexes, 'PolynomB', 0) 
`
works, giving the first element of PolynomB for the elements of ring with indexes in array_of_indexes.

`get_value_refpts(ring, array_of_indexes, 'Length', 0) 
`
instead fails, as Length is a scalar. 

the help of get_value_refpts says it: " ... **IF** attr_name is an array" 

I find this somehow strange that the user has to remember if the specific attribute is an array or not when using the command `get_value_refpts`. 

The error message is a python error and it is not very immediate. It could be caught and translated to a warning. For example: "For scalar attributes index is ignored". 

This action, would also recover a usage, similar to the matlab equivalent of this function `atgetfieldvalues`.

lfarv, 2022-09-05:
@simoneliuzzo:

Unlike in Matlab, scalar and numpy arrays are completely different python objects ! You indeed need to know what is the attribute. But I doubt that you would extract attribute values without knowing what they mean… 

Anyway, I very much prefer using a simple python command rather that using `get_value_refpts`: in your example, you would write
```python
b0=[elem.PolynomB[0] for elem in ring.select(array_of_indexes)]
```
or
```python
lengths=[elem.Length for elem in ring.select(array_of_indexes)]
```
It's more efficient, it works for integer or boolean indices (`get_value_refpts` also, of course) and it's self-explanatory for a python developer… The function does not bring anything, but it's a matter of taste.

swhite2401, 2025-07-23:
Solved in #959

closed 2025-07-23 by swhite2401