atcollab/at #502: Matlab: *QuantPass.c not working anymore

swhite2401, 2022-10-18:
@JonasKallestrup has found that the element-by-element photon emission does not work anymore.

After some testing it turns out that this problem is specific to matlab: particle are lost after they cross an element with `*QuantPass.c.`

The same tracking in pyAT works perfectly fine, I would therefore be tempted to blame the matlab `mex` command, and the way file are compiled in matlab. Any matlab expert have a good idea? @lfarv @lnadolski @carmignani @simoneliuzzo ?

A strange thing to note: adding a `printf()` statement in the passmethod magically fixes the issue.

There seem to be bugs in the passmethods:
-p_norm, NormL1, NormL2 probably need to be recomputed at each loop
-updating the r6[1] and r6[3] coordinates after the energy change does not seems right

These need to be confirmed and fixed as well.

swhite2401, 2022-10-18:
Below a python test script:

```
import at
import numpy
import matplotlib.pyplot as plt
from at.tracking import lattice_pass

path  = './'
filename = 'S28F.mat'
key = 'LOW_EMIT_RING_INJ'
latticef = path+filename
ring = at.load_lattice(latticef,key=key)
ring.radiation_on()

o6, _ = ring.find_orbit()
ncases = 5

xrad = []
for a in range(ncases):
    rout = numpy.squeeze(lattice_pass(ring, o6.copy(), refpts= range(60)))
    xrad.append(rout[0,:])

xquant = []
ring.radiation_off(cavity_pass='RFCavityPass')
rdip = ring.get_refpts(at.Dipole)
for d in ring[rdip]:
    d.PassMethod = 'BndMPoleSymplectic4QuantPass'
    d.Energy = ring.energy
    
for a in range(ncases):
    rout = numpy.squeeze(lattice_pass(ring, o6.copy(), refpts= range(60)))
    xquant.append(rout[0,:])

for i, xr, xq in zip(range(ncases),xrad,xquant):
    plt.subplot(211)    
    plt.plot(xr, label = 'Case #'+str(i))
    plt.subplot(212)    
    plt.plot(xq, label = 'Case #'+str(i))    
    
plt.subplot(211)    
plt.xlabel('Element index')
plt.ylabel('x [m]')
plt.legend()
plt.subplot(212)    
plt.xlabel('Element index')
plt.ylabel('x [m]')
plt.legend() 
plt.show()
```

swhite2401, 2022-10-18:
Below a matlab test script:

```
clear all;

v = load('./S28F.mat');
ring = v.LOW_EMIT_RING;
ring = atradon(ring);

ClosedOrbit = findorbit6(ring);

for a = 1:5
    Rout = linepass(ring,ClosedOrbit,1:60);% track through first 30 elements
    x_all_withoutQuant(:,a) = Rout(1,:);
    leg_withoutQuant{a} = ['#',num2str(a),'. Loss index: ',num2str(find(isnan(x_all_withoutQuant(:,a)),1,'first'))];
end

ring = atradoff(ring);
ring = atQuanton(ring);
for a = 1:5
    Rout = linepass(ring,ClosedOrbit,1:60);% track through first 30 elements
    x_all_withQuant(:,a) = Rout(1,:);
    leg_withQuant{a} = ['#',num2str(a),'. Loss index: ',num2str(find(isnan(x_all_withQuant(:,a)),1,'first'))];
end

%%
figure('pos',[10,10,1200,400])
subplot(1,2,1)
plot(x_all_withoutQuant)
set(gca,'FontSize',14)
xlabel('Element index')
ylabel('x [m]')
legend(leg_withoutQuant,'Location','northwest')
title('BndMPoleSymplectic4RadPass','FontSize',13)

subplot(1,2,2)
plot(x_all_withQuant)
set(gca,'FontSize',14)
xlabel('Element index')
ylabel('x [m]')
legend(leg_withQuant,'Location','northwest')
title('BndMPoleSymplectic4QuantPass','FontSize',13)

```

lfarv, 2022-10-18:
@swhite2401 : can you post the `atQuanton.m` function used in you Matlab test ? Thanks !

swhite2401, 2022-10-18:
There it is (strange that it was not added to the distribution..., there is the same for OFF):
```

function [ring,radelems,cavities,energy]=atQuanton(ring1,varargin)
%ATQuantON			switches RF and quantum radiation on
%
%RING2=ATQuantON(RING,CAVIPASS,BENDPASS,QUADPASS)
%   Changes passmethods to get RF cavity acceleration and radiation
%   damping. ATQuantON also sets the "Energy" field on the modified elements,
%   looking for the machine energy in:
%       1) 1st 'RingParam' element
%       2) 1st 'RFCavity' element
%       3) field "E0" of the global variable "GLOBVAL"
%
%RING:		initial AT structure
%CAVIPASS:	pass method for cavities (default ThinCavityPass)
%           '' makes no change,
%BENDPASS:	pass method for bending magnets. Special values:
%           '' makes no change,
%           'auto' wille substitute 'Pass' with 'QuantPass' in any method
%           (default: 'auto')
%QUADPASS:	pass method for quadrupoles
%           '' makes no change,
%           'auto' wille substitute 'Pass' with 'QuantPass' in any method
%           (default: '')
%
%[RING2,RADINDEX,CAVINDEX,ENERGY]=ATQuantON(...)
%           returns the index of radiative elements and cavities + energy

[cavipass,bendpass,quadpass]=parseargs({'CavityPass','auto',''},varargin);

ring=ring1;

energy=atenergy(ring);
if ~isempty(cavipass)
    cavities=atgetcells(ring,'Frequency');
    if any(cavities)
        ring(cavities)=changepass(ring(cavities),cavipass,energy);
    end
else
    cavities=false(size(ring));
end

if ~isempty(bendpass)
    isdipole=@(elem,bangle) bangle~=0;
    dipoles=atgetcells(ring,'BendingAngle',isdipole);
    if any(dipoles) > 0
        ring(dipoles)=changepass(ring(dipoles),bendpass,energy,20);
    end
else
    dipoles=false(size(ring));
end

if ~isempty(quadpass)
    isquadrupole=@(elem,polyb) length(polyb) >= 2 && polyb(2)~=0;
    quadrupoles=atgetcells(ring,'PolynomB',isquadrupole) & ~dipoles;
    if any(quadrupoles) > 0
        ring(quadrupoles)=changepass(ring(quadrupoles),quadpass,energy);
    end
else
    quadrupoles=false(size(ring));
end

radelems=dipoles|quadrupoles;

if any(cavities)
    disp(['Cavities located at position ' num2str(find(cavities)')]);
else
    disp('No cavity');
end
disp([num2str(sum(radelems)) ' elements switched to include quantum radiation']);

    function newline=changepass(line,newpass,nrj,varargin)
    if strcmp(newpass,'auto')
        passlist=atgetfieldvalues(line,'PassMethod');
        ok=cellfun(@(psm) isempty(strfind(psm,'QuantPass')),passlist);
        passlist(ok)=strrep(passlist(ok),'Pass','QuantPass');
    else
        passlist=repmat({newpass},size(line));
    end
    newline=cellfun(@newelem,line,passlist,'UniformOutput',false);

        function elem=newelem(elem,newpass)
            elem.PassMethod=newpass;
            elem.Energy=nrj;
            if(size(varargin)>0)
            elem.NumIntSteps=varargin{1};
            end
        end
    end

end
```

lfarv, 2022-10-18:
@swhite2401, @JonasKallestrup: for me, the test function provided by @swhite2401 works on both Matlab and python. Can you give more details on what you observe: which platform, Matlab version, with or without OpenMP.

Concerning the bugs:
> There seem to be bugs in the passmethods:
> -p_norm, NormL1, NormL2 probably need to be recomputed at each loop

Obviously this one must be corrected.

> -updating the r6[1] and r6[3] coordinates after the energy change does not seems right

Here I don't know yet…

Now for the missing` atQuanton/off`, it needs some upgrade to be included, and I'd like to see first if it can be integrated in the recently updated `atenable_6d` !

swhite2401, 2022-10-19:
For me, the platform is unix:

  Hostname:  slurm-nice-devel1904
  IP:        - 160.103.198.13
             - 160.103.228.43
  Hardware:  x86_64
  System:    Linux 5.4.0-122-generic (Ubuntu 20.04.4 LTS)

The matlab version is 2020a, I ran atmexall several time and also tried with matlab 2018a on the same platform

swhite2401, 2022-10-19:
For the bugs, it is easy to fix we can think about it later, I agree one is obvious the other not as much...
Integration of `quanton/off` in `enable_6d` would be great!
I had done a branch for this in python since CERN colleagues were using this feature but never made a PR, it is completely outdated now with the introduction of `enable_6d`.
The tricky part is to avoid conflicts between `RadPass` and `QuantPass`

lfarv, 2022-10-20:
@swhite2401 : I can reproduce the problem on "grappa": very strange indeed! So I can start looking at the problem.

To switch quantum radiation on, the easiest way at the moment is:
```Matlab
ring = atdisable_6d(ring);      % For safety
ring = atenable_6d(ring,allpass='',cavipass='auto',bendpass='BndMPoleSymplectic4QuantPass');
```
But unfortunately, at `disable_6d` ignores the radiating elements: to be corrected!

swhite2401, 2022-10-20:
> @swhite2401 : I can reproduce the problem on "grappa": very strange indeed! So I can start looking at the problem.
>
> To switch quantum radiation on, the easiest way at the moment is:
>
> ```matlab
> ring [...]

Not only bends are concerned, maybe we could have a separate function that correction positions the keyword arguments to have the whole ring as *QunatPass?

swhite2401, 2022-10-20:
Ah many bugs! I think we could treat all these in dedicated PR, once we know what is wrong with the build on ubuntu...

lfarv, 2022-10-20:
I suspect that `BndMPoleSymplectic4QuantPass` is really wrong; Image the following test:
- run 1000 samples of tracking S28F over 1 turn, without RF cavity but with quantum radiation in dipoles, and plot an histogram of momentum loss: the loss distribution is compatible with a Poisson distribution, with a mean of 1.13e-3
- run the same tracking but with "classical" radiation (`BndMPoleSymplectic4RadPass`). The 1000 samples are identical (of course !) with a loss of 0.4228e-3, compatible with the analytical result.

[image]

**Conclusion: the quantum radiation gives an energy loss more than twice the expected value.**

A full review of the code is necessary. Is there a reference paper related to this code?

swhite2401, 2022-10-20:
> **Conclusion: the quantum radiation gives an energy loss more than twice the expected value.**
> 
> A full review of the code is necessary. Is there a reference paper related to this code?

I am very surprised about this results, this had been fully benchmarked (energy loss, equilibrium emittance, etc..) when it was implemented (in 2014 I think) and gave excellent results. @carmignani fixed the multipoles after that I think and did further check but such an error was never reported.

There is no reference for this except for a presentation that I gave in 2014 with all the formulas, I can try find it back.

That being said, there seem to be many bugs here and there and I agree that a full review is required.
Did you understand the problem reported by @JonasKallestrup ?

swhite2401, 2022-10-20:
Here is the presentation, everything looked fine at the time...

[AT_SR.pptx](https://github.com/atcollab/at/files/9832297/AT_SR.pptx)

lfarv, 2022-10-20:
Hi @swhite2401: thanks for the ppt !

No result yet concerning the Ubuntu bug: I was just trying to identify possible outliers in the energy kick, using the histogram of energy loss, when I found this problematic result.

swhite2401, 2022-10-20:
@lfarv, same exercise in python, looks more reasonable... there is something clearly odd with the matlab build...

[image]

JonasKallestrup, 2022-10-21:
> Hi @swhite2401: thanks for the ppt !
> 
> No result yet concerning the Ubuntu bug: I was just trying to identify possible outliers in the energy kick, using the histogram of energy loss, when I found this problematic result.

I don't think it is an "Ubuntu bug" since I have the issue on Windows.

swhite2401, 2022-10-21:
Thanks @JonasKallestrup, I still have no idea what is going on though...

swhite2401, 2022-10-21:
It has a name: https://en.wikipedia.org/wiki/Heisenbug

swhite2401, 2022-10-21:
Bug found:
in `atquantlib.c`, the static variable `InitSeed` has to be declared in the function `drand() `not at the top level

I will start a new PR, we can use it the review the rest of the code.

swhite2401, 2022-10-21:
> Bug found: in `atquantlib.c`, the static variable `InitSeed` has to be declared in the function `drand() `not at the top level
> 
> I will start a new PR, we can use it the review the rest of the code.

Sorry... just a lucky case... I keep looking

swhite2401, 2022-10-21:
Ok so I think I have found the issue, the variable `lb` in `atquantlib.get_Energy()` was set to 0.02 in the latest commit in 2017.
This value is outside the look up table: the table search was return 0 and then the interpolation fails. It appears this was the source of this erratic bug. I have no idea why it worked (or seemed to work) in some cases and not in other...

Setting the value back to 0.21 seems to fix the issue. @carmignani do you remember why you had set such a low value? The commit says "new version with bug fix" but it looks like it introduced a bug instead...

@lfarv, @JonasKallestrup, could you please confirm on your side that it is ok with this change? Should I create a branch for testing?

swhite2401, 2022-10-21:
Ah interesting! I have absolutely no memories of this! Well this was obviously a mistake...

lfarv, 2022-10-21:
Hi all
Setting `lb` to 0.21 gives much better results: here is the new plot

[image]

Mean dp/p for 1000 "QuantPass" samples: -4.2532e-04
dp/p for "RadPass": -4.2274e-04

That makes sense !

Note that on MacOS, the wrong Matlab results were identical on python…
OK for a test branch.

JonasKallestrup, 2022-10-21:
I'll parrot what @lfarv said: setting lb=0.21 fixes the problem and provides reasonable results for energy loss per turn.

swhite2401, 2022-10-21:
Ok great, Ill set up a branch and also fix the other bugs that were found in this discussion.

lfarv, 2022-10-31:
Fixed in #504

closed 2022-10-31 by lfarv