atcollab/at #514: pyat 0.3.0 sets radiation_on sets PassMethod to non-existing RFCavityRadPass

oscarxblanco, 2022-11-09:
Dear all,

I'm using pyat version 0.3.0 and I see that the function radiation_on with cavity_pass='auto' sets the  method to "RFCavityRadPass" which is not a valid Pass method.

`ring.radiation_on(cavity_pass='auto') # BUG :: Cavity with non-existing PassMethod RFCavityRadPass`

However,  setting explicitely RFCavityPass works

`ring.radiation_on(cavity_pass='RFCavityPass')`

Was it the intended usage ? Or should I update my pyat version ?

Best regards,

o

swhite2401, 2022-11-10:
Hello @oscarxblanco,
I have not been able to reproduce this issue, I did pull the latest version from the master to do the test.
How do you build your cavity element?

lfarv, 2022-11-10:
Hello @oscarxblanco:
I'm also not able to reproduce your problem. Can you post your lattice ?

oscarxblanco, 2022-11-10:
dear all,

I mistakenly used pyat 0.2.1 instead of 0.3.0;
It was not updated in the virtual environment I was using, and I was not able to confirm it because at.__version__ did not work.

Just after your messages I suspected of a wrong pyat version and I went to launch pip install accelerator-toolbox --upgrade blindly. It confirmed the unistallation of 0.2.1 and the installation of 0.3.0.

With pyat 0.3.0 I have no errors when setting cavitypass='auto', and at.__version__ shows the expected 0.3.0.

Thank you for your quick response, from my side this issue could be closed.

o

closed 2022-11-15 by swhite2401