atcollab/at #518: pip install -e not working with MPI

swhite2401, 2022-11-15:
The `-e` option returns an error when trying to build pyAT with MPI,
Command used:
`pip install --config-settings mpi=1 ".[mpi]" -e .`

Is this not allowed?

lfarv, 2022-11-17:
It works for me… What error do you get?
Note that if you have already compiled without MPI, the compilation will not occur again unless you modify a .c file. The easiest way to force a new compilation is to remove the `build` directory: `rm -rf build`.

swhite2401, 2022-11-17:
Here it is

[image]

swhite2401, 2022-11-17:
I get this issue on the slurm-nice-devel machine with a python 3.8 virtualenv

lfarv, 2022-11-19:
I tried the same, and for me it works:

```
slurm-nice-devel1904:at $ # create a virtual environment
slurm-nice-devel1904:at $ 
slurm-nice-devel1904:at $ python3 -m venv ~/python/sys38

slurm-nice-devel1904:at $ # activate the virtual environment
slurm-nice-devel1904:at $ 
slurm-nice-devel1904:at $ source ~/python/sys38/bin/activate

(sys38) slurm-nice-devel1904:at $ # Update pip
(sys38) slurm-nice-devel1904:at $ 
(sys38) slurm-nice-devel1904:at $ pip install -U pip setuptools
Collecting pip
  Using cached pip-22.3.1-py3-none-any.whl (2.1 MB)
Collecting setuptools
  Using cached setuptools-65.6.0-py3-none-any.whl (1.2 MB)
Installing collected packages: pip, setuptools
  Attempting uninstall: pip
    Found existing installation: pip 20.0.2
    Uninstalling pip-20.0.2:
      Successfully uninstalled pip-20.0.2
  Attempting uninstall: setuptools
    Found existing installation: setuptools 44.0.0
    Uninstalling setuptools-44.0.0:
      Successfully uninstalled setuptools-44.0.0
Successfully installed pip-22.3.1 setuptools-65.6.0

(sys38) slurm-nice-devel1904:at $ # Install
(sys38) slurm-nice-devel1904:at $ 
(sys38) slurm-nice-devel1904:at $ pip install --config-settings mpi=1 -e .[mpi]
Obtaining file:///mnt/multipath-shares/machfs/laurent/dev/libraries/at
  Installing build dependencies ... done
  Checking if build backend supports build_editable ... done
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
  Preparing metadata (pyproject.toml) ... done
Collecting importlib-resources
  Using cached importlib_resources-5.10.0-py3-none-any.whl (34 kB)
Collecting scipy>=0.16
  Using cached scipy-1.9.3-cp38-cp38-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (33.8 MB)
Collecting numpy>=1.16.6
  Using cached numpy-1.23.4-cp38-cp38-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (17.1 MB)
Collecting mpi4py
  Using cached mpi4py-3.1.4-cp38-cp38-linux_x86_64.whl
Collecting zipp>=3.1.0
  Using cached zipp-3.10.0-py3-none-any.whl (6.2 kB)
Installing collected packages: zipp, numpy, mpi4py, scipy, importlib-resources, accelerator-toolbox
  Running setup.py develop for accelerator-toolbox
Successfully installed accelerator-toolbox-0.3.1.dev11+gc20694c5 importlib-resources-5.10.0 mpi4py-3.1.4 numpy-1.23.4 scipy-1.9.3 zipp-3.10.0
(sys38) slurm-nice-devel1904:at $ 
```

Try rebuilding your virtual environment or updating it

swhite2401, 2022-11-21:
Ok my bad it works for me as well, I had written:
`pip install --config-settings mpi=1 ".[mpi]" -e .`

instead of
`pip install --config-settings mpi=1 -e ".[mpi]"`

It seems that the quotes around `.[mpi]` are needed (you did not use them in fact) could you confirm?

closed 2022-11-22 by swhite2401