atcollab/at #522: atidtable_dat (IdTablePass) in pyat

oscarxblanco, 2022-11-18:
Dear all,

is there a pyat function to create an element from a Radia magnetic field map (pass method 'IdTablePass') ?

In matlab AT it is called atidetable_dat, however, I haven't been able to find an equivalent in pyat.
The 'IdTablePass' is in the integrators folder but I see no usage of it.

Best regards,
o

lcarver, 2022-11-30:
Hello,
Have you tried generating a general element?
```

from at.elements import Element
elem = Element('name', PassMethod='IdTablePass', kwargs*)
```

where kwargs needs to provide all of the correct arguments for the passmethod.

oscarxblanco, 2022-11-30:
Dear @lcarver ,
thank you for pointing me out to the general element function. I had a quick view and it seems that it does not have all the arguments required by the field map tracking. 

Here is a list of the element arguments definitions in `elements.py` :

at/lattice/elements.py 
231     _BUILD_ATTRIBUTES = ['FamName']                                             
232     _conversions = dict(FamName=str, PassMethod=str, Length=float,              
233                         R1=_array66, R2=_array66,                               
234                         T1=lambda v: _array(v, (6,)),                           
235                         T2=lambda v: _array(v, (6,)),                           
236                         RApertures=lambda v: _array(v, (4,)),                   
237                         EApertures=lambda v: _array(v, (2,)),                   
238                         KickAngle=lambda v: _array(v, (2,)),                    
239                         PolynomB=_array, PolynomA=_array,                       
240                         BendingAngle=float,                                     
241                         MaxOrder=int, NumIntSteps=int,                          
242                         Energy=float,                                           
243                         )

While in matlab I get :
[image]

Do these differences have an impact on the tracking ?

lcarver, 2022-12-12:
From looking at IDtablepass, the required arguments are: Length, xkick, ykick, xtable, ytable, Nslice

So if you use 
`elem = Element('name', PassMethod='IdTablePass', xkick=x0, ykick=x1, xtable=x2, ytable=x3, Nslice=x4)
`
then they will be passed to the C passmethod and it should work (assuming your input is ok).

swhite2401, 2022-12-12:
Hello @oscarxblanco, you are also more than welcome to propose a python implementation of this `at` element, we can help you during the review process.

oscarxblanco, 2022-12-15:
Thank you very much, I'll work on it.

oscarxblanco, 2023-01-18:
Dear all,

I have finished a first version of an InsertionDevice class that loads a field map file, and the needed parameters.
At the moment I have two problems : 
1) the lattice.insert function does not copy the Energy parameter, but, it is required to normalize the field map.
2) I see a discrepancy in the tune variation wrt matlab for the same field map and lattice (approx delta_nuy= 4.4e-3 in matlab at, and almost zero in pyat). Is the IdTablePass method valid when using the function get_optics ?

I might have a problem in the class definition, so, I am sharing this preliminary version in a zip file, with a lattice, a field map, and and a jupyter script for a quick test.

InsertionDevice.py  is the class definition
lat_soleil.m is a test lattice with a long enough drift.
u20_g45mm_kicks_2022nov10.txt is a field map example.
test_insertID.ipynb is the jupyter test script.

[image]

[pyatInsertionDevice.zip](https://github.com/atcollab/at/files/10448866/pyatInsertionDevice.zip)

swhite2401, 2023-01-19:
@oscarxblanco , I''l take a look and let you know. Going very quickly through it I can see that you did not implement the `super` command in your `__init__()` function.
I do not know whether this is the problem but I think it should be there.

swhite2401, 2023-01-19:
Oops sorry, my scrolling tool was misfunctioning! Forget my last comment. I'll take a look

oscarxblanco, 2023-01-19:
No problem. 

I attach a short script I used in matlab to calculate  the tune difference, in case it is useful.

(Please remove the .txt extention at the end of the filename) 
[check_ID_inthe_SOLEILring.m.txt](https://github.com/atcollab/at/files/10455861/check_ID_inthe_SOLEILring.m.txt)

o

oscarxblanco, 2023-01-19:
Dear all, 

I think I found my error. I did a fortran memory sorted copy too early.
Now it looks like this and it seems to work. I attach to this message the new InsertionDevice class.

    def sorted_table(self, table_in, sorted_index, order_axis):
        # np.asfortranarray makes a copy of contiguous memory positions
        table_out = np.copy(table_in);
        for i,iis in zip(range(len(sorted_index)), sorted_index):
            if order_axis == 'col': table_out[:,i] = table_in[ : ,iis]
            if order_axis == 'row': table_out[i,:] = table_in[iis, : ]
        table_out2 = np.asfortranarray(table_out)
        return table_out2

Best regards,

o

[image]
[InsertionDevice.py.txt](https://github.com/atcollab/at/files/10457248/InsertionDevice.py.txt)

oscarxblanco, 2023-01-24:
I just noticed that the InsertionDevice I created covers the only the kickmap method mentioned in :
issue  https://github.com/atcollab/at/issues/1

P.S.
I did a frequency map analysis of a lattice with an InsertionDevice in at matlab and in pyat. From the plot I see little-to-no difference.

swhite2401, 2023-01-25:
Hello @oscarxblanco it looks like you have done some very nice developments (this and your FMA tool). I think the best is that you create pull requests with myself @lfarv @lcarver @carmignani and @simoneliuzzo as reviewers.
This way we can start the code review and eventually integrate them in the distribution.

oscarxblanco, 2023-01-25:
ok, Great !
Should it be another branch ? or a fork ?

o

lcarver, 2023-01-25:
Hello,
the way to do it is...
first make sure you have saved somewhere outside of the pyat repository all of your changes
checkout the master branch of pyat and do a `git pull` 
create a new branch with `git checkout -b 'name_of_branch'`
`git push` will then register this new branch
then commit and push all of your changes and modifications
then find your branch online here and create a pull request 

We are then able to checkout your branch and make some proposals based on our tests

swhite2401, 2023-01-25:
I suggest you make one branch/PR for FMA and one for idtablepass

oscarxblanco, 2023-02-01:
Good, 
I started with a new branch,  but, at the moment it seems I need higher permissions as I got the following error :
```
$ git push --set-upstream origin PR
ERROR: Permission to atcollab/at.git denied to oscarxblanco.
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
```

swhite2401, 2023-02-01:
I have invited you to be a member of the project, this should give you the rights to create pull request, let me know if this works.

oscarxblanco, 2023-02-01:
Thank you indeed,

I confirm that I am able to push, and after some checks on my side I will do the pull request as suggested.

Best regards,
o

swhite2401, 2023-02-23:
Handled in PR #558

closed 2023-02-23 by swhite2401