atcollab/at #552: patpass number of cpu in a pool

oscarxblanco, 2023-01-23:
Dear all,

is there a way to know the number of cpus used in a patpass pool ?
it seems you that the number of cpus asked and given from a pool could be different.

Best regards,
o

swhite2401, 2023-01-31:
Hello @oscarxblanco,

If no users input is given, the poolsize is defined as below:
`pool_size = min(len(r_in[0]), multiprocessing.cpu_count(), DConstant.patpass_poolsize)`
which is the minimum between, number of particles tracked, available CPUs or global option `DConstant.patpass_poolsize`.

If you set the `pool_size` argument, then this number is used, however if the number of particles is smaller than the `pool_size` `patpass` will spawn only the needed number of threads which is in this case equal to the number of particles.

oscarxblanco, 2023-02-01:
Great, thank you !

Is there anyway to confirm the value of the poolsize once created ?

Best regards,
o

swhite2401, 2023-02-01:
Sorry having a closer look at the code to answer your question, I realized I gave you wrong information: if you set the argument `pool_size ` of `patpass` this number is used for the pool generation regardless of the number of particles.
If you want to control the number of processes this is probably what you want to use.

However if the number of particles is smaller than `pool_size`  you will have empty processes, I don't know whether this is a problem for you.