atcollab/at #592: AT does not compile with OpenMP=1

swhite2401, 2023-04-06:
The C code including random number (QuantdiffPass, *QuantPass, etc...) generation does not compile anymore if OpenMP is activate below the error:

[image]

It looks like the random generator has to be added to the shared variables. @lfarv do you confirm?

swhite2401, 2023-04-06:
In fact adding rng to the shared variables allows to compile correctly, I have test if it runs...

lfarv, 2023-04-07:
The `rng` variable must indeed be added to the shared variables. I then compiles and runs, but there is another problem: the average energy loss from quantum radiation is ~10% too high with OpenMP.

I'm looking at that.

lfarv, 2023-04-07:
Update: the test is done looking at the energy loss over one turn of S28F, with 1000 particles and `BndMPoleSymplectic4QuantPass` compared to `BndMPoleSymplectic4RadPass`.

The results are in excellent agreement without OpenMP and differ by ~10% with OpenMP

lfarv, 2023-04-07:
### Results without OpenMP

[image]
[image]

Here is the notebook used for the test (remove the .txt extension):
[testquantum.ipynb.txt](https://github.com/atcollab/at/files/11178428/testquantum.ipynb.txt)

lfarv, 2023-04-07:
### Results with OpenMP:
[image]
[image]

I do not understand what may happen with the random generator… It would be interesting to do the same test with MPI.

For the time being, it seems necessary to disable OpenMP for the 2 passmethods `BndMPoleSymplectic4QuantPass`, `StrMPoleSymplectic4QuantPass` and `QuantDiffPass`. What do you think ?

swhite2401, 2023-04-11:
Yes I agree, it has to be disabled. I will check that results are correct with MPI