atcollab/at #606: include fast_ring generation from a list of machine parameters (tunes, chromaticities, ac, U0, etc)

lcarver, 2023-05-24:


swhite2401, 2023-05-25:
@lfarv this is done in other collective effects codes using just damping times + equilibrium emittances (exponential decay + white noise which amplitude depends on the distance w.r.t the target)

For the rest I think it is straight forward as long as you assume uncoupled motion.

I have somewhere an implementation in matlab that I used before we introduced the fastring.

lcarver, 2023-05-26:
So for some more information...

I want to make a function that can provide some simple longitudinal and transverse elements for a given list of global machine parameters i.e.

fring = gen_uncoupled_ring(Qx, Qy, Qpx, Qpy, axx, axy, ayy, Vrf, phi_s, U0, alphac, emittances, damping_times)

And then fring would be composed of:

a 6x6 tracking element (uncoupled and without radiation damping)
a nonlinear element (for chromaticity and amplitude detuning)
an rf cavity element
a damping element to include the radiation damping and equilibrium emittance (as Simon said, to see the pyheadtail implementation see here: https://github.com/PyCOMPLETE/PyHEADTAIL/blob/master/PyHEADTAIL/radiation/radiation.py)

If we have this, I can (for example) take a table of parameters for any other machine, and make some simulations. I don't need to have the full lattice model.

lfarv, 2023-08-29:
This seems to be answered in #643. Can we close the issue?

lcarver, 2023-08-31:
thanks to all.

closed 2023-08-31 by lcarver