atcollab/at #630: atm2html not working

lnadolski, 2023-07-23:
Dear @lfarv 

Is the function atm2html the only way to get HTML documentation for AT?
The function calls for a template frame-at that is not part of the main repository.

It would be nice to have something similar to pyAT doc.
Is there any way to reach this goal?

Best regards,

Laurent.

lfarv, 2023-07-27:
Yes. This was relying on an external service with which I had problems. The options for the Matlab documentation are now:

1. Integration in the Matlab help system: at the bottom of the Matlab help window, a section "Supplemental Software" gives access to "AT: Accelerator Toolbox". Did you have a look?
2. Sphinx: this is used for the PyAT help, but can also work with Matlab files. The problem is that it expects help sections formatted like the python docstrings, rather than in the "standard" Matlab style. This option requires rewriting most of the documentation of functions.

Both options, as `atm2html` did, can automatically generate the documentation for all functions. But the `atmat` directory is not organised well enough to give a coherent tree of function names. A first step would be to reorganise `atmat` in a more coherent tree, and hide the private functions which should not appear. For the time being, I put in the `developer` section some scripts to manually tune the automatic generation of the documentation.

In addition, there is the possibility to put documentation in the form of Matlab notebooks. Please have a look at the few examples in the Matlab help viewer.

I am at the moment in favour of using the Matlab help system, rather than Sphinx. Look at the very preliminary state in the Matlab help viewer, and give your comments !

lnadolski, 2023-07-28:
Dear @lfarv,

Thanks for reminding me the option 1. I had a quick look.
I will continue later.

swhite2401, 2025-07-23:
Solved in #951