atcollab/at #637: Closed orbit in momentum aperture

oscarxblanco, 2023-07-28:
Dear all,

in `get_momentum_acceptance` pyat calculates the closed orbit on each reference point.
For a lattice with errors the closed orbit guess around zeros might not converge.

It would be better to either calculate the closed orbit at the start of the ring and propagate the coordinates througth out the accelerator up to the reference point, or to store the orbit in all reference points and use it later on in the routine.

I have tried the second approach by using the parameter offset in `get_acceptance` to pass the closed orbit, however, it seems to consider a single 6D point offset.

Is there any way to pass the closed orbit to momentum aperture ? or is offset able to work with multiple reference points ?

Best regards,
o

swhite2401, 2023-07-31:
@oscarxblanco `get_1d_acceptance` that is used in `get_momentum_acceptance` has no offset parameter.
In any case you are correct, the offset is not correctly handled in the presence of multiple `refpts` so this would need to be improved.

I can propose the following changes:
-adapt offset in `get_acceptance` to handle multiple `refpts`
-introduce offset in `get_1d_acceptance`

In this way you should be able to pre-compute the closed orbit at your observations points and pass it to the function function through the offset argument.

Would that work for you?

oscarxblanco, 2023-07-31:
Yes, this sounds great

closed 2023-08-04 by oscarxblanco