atcollab/at #646: internal_epass not working

TeresiaOlsson, 2023-08-25:
Hi,

The function internal_epass doesn't work for my lattice, but I haven't managed to figure out why. Can someone perhaps test it and see if you get the same error as me?

I have attached my lattice file (you need to change from .txt
[M_H6BA_38_1_1_mod.txt](https://github.com/atcollab/at/files/12439414/M_H6BA_38_1_1_mod.txt)
 to .lat) and the following is a simple example of what I get:

ring = at.load_tracy("M_H6BA_38_1_1_mod.lat",harmonic_number=934)
ring.harmonic_number = 934
print(at.envelope_parameters(ring))

I get the following error:

AtWarning: Harmonic number changed, resetting fillpattern to default (single bunch)
  warn(AtWarning('Harmonic number changed, resetting fillpattern to '
Traceback (most recent call last):

  File ~/Software/miniconda3/envs/pyat-dev/lib/python3.11/site-packages/spyder_kernels/py3compat.py:356 in compat_exec
    exec(code, globals, locals)

  File ~/Documents/Johan-collective-effects-benchmark/Diamond-II-pyAT/diffusion-matrix/diffusion_benchmark.py:21
    print(at.envelope_parameters(ring))

  File ~/Documents/git-repos/at-fork/at/pyat/at/physics/ring_parameters.py:193 in envelope_parameters
    emit0, beamdata, emit = ohmi_envelope(ring, orbit=orbit,

  File ~/Documents/git-repos/at-fork/at/pyat/at/lattice/utils.py:212 in wrapper
    return func(ring, *args, **kwargs)

  File ~/Documents/git-repos/at-fork/at/pyat/at/physics/radiation.py:168 in ohmi_envelope
    bbcum, orbs = _dmatr(ring, orbit=orbit, keep_lattice=keep_lattice)

  File ~/Documents/git-repos/at-fork/at/pyat/at/physics/radiation.py:65 in _dmatr
    bbcum = numpy.stack(list(_cumulb(zip(ring, orbs, bb))), axis=0)

  File ~/Documents/git-repos/at-fork/at/pyat/at/physics/radiation.py:46 in _cumulb
    m = find_elem_m66(el, orbin, energy=energy, particle=ring.particle)

  File ~/Documents/git-repos/at-fork/at/pyat/at/physics/matrix.py:207 in find_elem_m66
    internal_epass(elem, in_mat, **kwargs)

  File ~/Documents/git-repos/at-fork/at/pyat/at/tracking/utils.py:86 in wrapper
    return func(lattice, r_in, *args, **kwargs)

  File ~/Documents/git-repos/at-fork/at/pyat/at/tracking/track.py:61 in _element_pass
    return _elempass(element, r_in, **kwargs)

TypeError: argument 1 must be Element, not Marker

Best regards,

Teresia

swhite2401, 2023-08-25:
Hi Teresia,

No issue on my side with an ubuntu machine and latest pyAT from master, here is the output to your script. Results looks strange but I don't see any python error:

```
slurm-nice-devel2903:at_dev/pyat_jobs/misc % python ./diamond.py 
/mnt/multipath-shares/machfs/swhite/at_dev/at/pyat/at/lattice/lattice_object.py:844: AtWarning: Harmonic number changed, resetting fillpattern to default (single bunch)
  warn(AtWarning('Harmonic number changed, resetting fillpattern to '
  Frac. tunes (6D motion): [0.14976192 0.26468003 0.00250514]
                   Energy: 3.500000e+09 eV
       Energy loss / turn: 7.234665e+05 eV
          Mode emittances: [-0. -0. -0.]
Damping partition numbers: [3.97153819e+00 2.78014960e-02 6.60313991e-04]
            Damping times: [-2.41393555e+04 -3.44838897e+06 -1.45189067e+08] s
            Energy spread: 0
             Bunch length: 0 m
         Cavities voltage: 1420000.0 V
        Synchrotron phase: 2.60701 rd
    Synchrotron frequency: 1339.77 Hz
```

swhite2401, 2023-08-25:
Are you running the official pyAT version? It looks like you are using a fork

TeresiaOlsson, 2023-08-25:
Thank you, Simon. Then I will investigate if something is broken on my side. I think the results only look strange because in the example radiation is off. I have also run the latest release of the code instead of cloning the repository and then I get the same strange results unless I turn the radiation on.

I'm using a fork, but I forked it very recently so I'm just one commit behind. I will clone the official repository and see if I get the same error or not. In my fork I have made changes in the load_tracy function to also be able to load a thor-scsi file so perhaps I introduced some error when loading the lattice that caused this.

TeresiaOlsson, 2023-08-25:
I figured out the problem. Apparently it was caused by running an editable version of the repository in Spyder. Then I can for some reason only run a script once and the second time I get the error unless I restart the kernel. If I run it directly in a terminal it works fine. So the problem is likely how Spyder handles things.

closed 2023-08-25 by TeresiaOlsson