atcollab/at #647: Naming of the keyword arguments of simple_ring

lfarv, 2023-08-28:
The recent `simple_ring` function has some arguments in common with the existing `sigma_matrix` function. However, their names are slightly different. Apart from the mnemonic inconvenience, this prevents from using the same dictionary as input for both functions (I encountered the problem when trying to compare the "design" &Sigma;-matrix with the output of tracking a population through `simple_ring`: I need both functions).

So I would like to rename the `simple_ring` arguments coherently with those of `sigma_matrix`:
`beta_x` -> `betax`
`alpha_x` -> `alphax`
`emit_x` -> `emitx`
`sigma_dp` -> `espread`
For the last one, strictly speaking, it's indeed a momentum spread (not energy), so we could envisage adding an alias in `sigma_matrix` (`mspread` or even keeping `sigma_dp`). But I would not like to alias all the arguments…

Next: `simple_ring` could accept the same input as `sigma_matrix`:
- `twiss_in`, using `R` if present (this can define a coupled lattice), or `alpha` and `beta` (for uncoupled lattices),
- otherwise `alphax`, … as now
- one could even accept a &Sigma;-matrix, which can replace `alpha*`, `beta*`, `emit*`, `sigma_dp*`.

Third: we miss the Matlab simple ring: `atsimplering`

**Otherwise, simple_ring worked nicely in all my tests!**

lfarv, 2023-08-30:
Anybody willing to do something about that ? Or shall I do it ?

swhite2401, 2023-08-30:
I think @lcarver said he would do it, but most likely not for matlab...

lfarv, 2023-09-06:
> This is not clear to me.

The &Sigma;-matrix (beam matrix as returned by `ohmienvelope`, for instance) contains all the optics and emittance information. So, with &Sigma; and tunes, you can get the M<sub>66</sub> matrix and the emittances. Adding the damping times, you have everything.

Here, since the longitudinal motion is treated apart, I think we should use a 4D `twiss_in` input, generate a 4x4 transfer map and add the 44, 45, 54 and 55 terms as you do it now.

So to start from the R matrix in `twiss_in`, the easiest is first to build &Sigma; = &epsilon;<sub>1</sub>.R<sub>1</sub> + &epsilon;<sub>2</sub>.R<sub>2</sub>, then use `a_matrix` to get the A matrix, use the tunes to generate the rotation matrix S and finally build M<sub>44</sub> = A.S.A<sup>-1</sup>

If starting directly from &Sigma;, `a_matrix` will also return the needed emittances.

lcarver, 2023-09-20:
fixed in #655

closed 2023-09-20 by lcarver