atcollab/at #649: Beam Moments passmethod causes core dump

lcarver, 2023-08-29:
Hello,

Lets take a ring (for example full S28F.mat), and use it to generate a fast_ring (called fring).

If I do: 

```
bm = at.BeamMoments('bm')
fring.append(bm)
_ = at.lattice_pass(fring)
```

then everything works fine. But if I do

```
bm = at.BeamMoments('bm')
fring.append(bm)
sig = at.sigma_matrix(ring)
```
I get a core dump. 

@swhite2401 believes the issue comes from the recent lattice_track interface. For now, I can avoid the problem by ensuring that the beam moments is appended to the fring right before I track. He is looking into it.

swhite2401, 2023-08-29:
Yes I have found the problem, I am on it.

lcarver, 2023-09-20:
Fixed in #650

closed 2023-09-20 by lcarver