atcollab/at #653: atmaincavities bug if no cavity

lnadolski, 2023-09-05:
Dear @lfarv, 

Could we add error handling with a message if there is no cavity in the lattice?
Or should the function just return Nan?

For now, it makes an error.

Best regards,

Laurent.

lfarv, 2023-09-06:
@lnadolski: can you describe your error ? For me, it returns an empty index array, which is the expected behaviour. As it is done, it's the caller's responsibility to use `isempty` on the output and decide what to do. This is what is done in `atparamscan`: return `NaN` for RF voltage or frequency it there is no cavity.

This behaviour should be explicitly mentioned in the help, I'll look at that.

lnadolski, 2023-11-08:
@lfarv It was already corrected by you in https://github.com/atcollab/at/pull/656

closed 2023-11-08 by lnadolski