atcollab/at #666: In place behaviour of sbreak

lcarver, 2023-10-04:
Hello,

I encountered this behaviour during the tutorial. 

`ring = at.load_mat('./dba.mat', mat_key='RING')`

The following line does not do anything:
`ring_tmp.sbreak(0.2, at.Marker('hello')) `
and there is nothing here for `copy=True` or `copy=False`

at.sbreak does not exist so we cannot do:
`at.sbreak(ring_tmp, 0.2, at.Marker('hello')) `

Instead the working syntax was:
`ring_tmp = ring_tmp.sbreak(0.2, at.Marker('hello'))`

which seems a bit in contradiction to other lattice modification functions.

swhite2401, 2023-10-04:
sbreak modifies the lattice structure. An in place modification does not make much sense and therefore a new ring is returned.
sbreak is a member function of the class Lattice_object so it is only accessible through it and in my opinion this should remains like this.

The only improvement I can see would be to put an example usage in the help. Would that be suitable for you?

lcarver, 2023-10-04:
Yes I think it would make it clearer. I am happy to do the PR (within the next few days)