atcollab/at #676: SBEND ignores ByError in backwards compatibility call to atsbend

oscarxblanco, 2023-10-23:
Dear all,

I think there is a bug in the backwards compatibility call to atsbend in line 42 of sbend.m, or in the help:

Help output contains ByError :
https://github.com/atcollab/at/blob/66a7c92af28bed5900891b9c4fcbf6fae169e1a0/atintegrators/sbend.m#L6-L13

While sbend and atsbend calls do not have ByError:
https://github.com/atcollab/at/blob/66a7c92af28bed5900891b9c4fcbf6fae169e1a0/atintegrators/sbend.m#L1
https://github.com/atcollab/at/blob/66a7c92af28bed5900891b9c4fcbf6fae169e1a0/atintegrators/sbend.m#L42

The way this line is written seems to wrongly assign ByError to K. Or maybe ByError was never used and the help output needs to be fixed.

lfarv, 2023-10-24:
Here is the original send function as imported from the SLAC version:

```Matlab
function z=rbend(fname,L,A,A1,A2,K,method)
%BEND('FAMILYNAME',  Length[m], BendingAngle[rad], EntranceAngle[rad],
%	ExitAngle[rad], K, 'METHOD')
%	creates a new family in the FAMLIST - a structure with fields
%		FamName        	family name
%		Length         	length of the arc for an on-energy particle [m]
%		BendingAngle		total bending angle [rad]
%		EntranceAngle		[rad] (0 - for sector bends)
%		ExitAngle			[rad] (0 - for sector bends)
%		ByError				error in the dipole field relative to the design value 
%		K						quadrupole K-value for combined funtion bends
%		PassMethod        name of the function to use for tracking
% returns assigned address in the FAMLIST that is uniquely identifies
% the family

ElemData.FamName = fname;  % add check for identical family names
ElemData.Length			= L;
ElemData.MaxOrder			= 3;
ElemData.NumIntSteps 	= 10;
ElemData.BendingAngle  	= A;
ElemData.EntranceAngle 	= A1;
ElemData.ExitAngle     	= A2;
ElemData.ByError     	= 0;
ElemData.K      			= K;

ElemData.R1 = diag(ones(6,1));
ElemData.R2 = diag(ones(6,1));
ElemData.T1 = zeros(1,6);
ElemData.T2 = zeros(1,6);

ElemData.PolynomA			= [0 0 0 0];	 
ElemData.PolynomB			= [0 K 0 0]; 

ElemData.PassMethod 		= method;

global FAMLIST
z = length(FAMLIST)+1; % number of declare families including this one
FAMLIST{z}.FamName = fname;
FAMLIST{z}.NumKids = 0;
FAMLIST{z}.KidsList= [];
FAMLIST{z}.ElemData= ElemData;
```

`ByError` is indeed in the help as it is now, but it is not in the argument list, and it is set to zero. Then, I could not find a place where this attribute would be used. So:
- the present compatibility function behaves indeed like the original one, except that it does not set a `ByError` (useless?) attribute with a zero value.
- Unless we find a place where this attribute is used, I would just correct the help section (which in addition documents a `BEND` function while its name is `SBEND`)

oscarxblanco, 2023-10-24:
Dear @lfarv ,

I think that correcting the help message is enough. Thank you for checking the original function.

closed 2023-11-08 by oscarxblanco