atcollab/at #695: pyAT 'Multipole has no attribute PolynomB' exception for loading multipoles from a .mat file with both PolynomB and K value defined

space-budgie, 2023-11-12:
### Issue
While trying to load a `.mat` lattice file using `at.load.matfile.load_mat`, multipole magnets that have both a `PolynomB` and a `K` value defined raise an error:

```Python
AttributeError: In element QFEND, parameter K: 'Multipole' object has no attribute 'PolynomB'
```
This was tested for Python accelerator toolbox version 0.4.0.

Here, QFEND is defined with with the Matlab code
```matlab
QFEND 	= atmultipole('QFEND' , 0.250000, [0 0 0 0], [0  3.653849 0 0], 'StrMPoleSymplectic4Pass','MaxOrder',2);  
QFEND.K   = QFEND.PolynomB(2);
```
### Potential cause
I believe that the issue lies in the function call

https://github.com/atcollab/at/blob/1e32037159aef2122d0fc0952c245acf129fe7cc/pyat/at/lattice/elements.py#L628

where passing in `K` as part of `**kwargs` causes a call to

https://github.com/atcollab/at/blob/1e32037159aef2122d0fc0952c245acf129fe7cc/pyat/at/lattice/elements.py#L704-L706

but `PolynomB` is only defined later in

https://github.com/atcollab/at/blob/1e32037159aef2122d0fc0952c245acf129fe7cc/pyat/at/lattice/elements.py#L634

### (Temporary) fix
I managed to bypass the issue by changing the lines
https://github.com/atcollab/at/blob/1e32037159aef2122d0fc0952c245acf129fe7cc/pyat/at/lattice/elements.py#L628-L634
to
```Python
        # Set MaxOrder while PolynomA and PolynomB are not set yet
        super(ThinMultipole, self).__setattr__('MaxOrder', maxorder)
        # Adjust polynom lengths and set them
        len_ab = max(self.MaxOrder + 1, len_a, len_b)
        self.PolynomA = lengthen(poly_a, len_ab - len_a)
        self.PolynomB = lengthen(poly_b, len_ab - len_b)
        super(ThinMultipole, self).__init__(family_name, **kwargs)
```
so that `PolynomB` is defined prior to the potential call to the setter. 

However, I have not tested if this breaks anything else in the code.

T-Nicholls, 2025-06-26:
Hi @TeresiaOlsson,
Yes, I got distracted by other things and forgot to circle back to this.
I'll implement the solution discussed on #895 and make a PR for it in the next couple of days

TeresiaOlsson, 2025-06-26:
Thank you. That is really great! :)

T-Nicholls, 2025-06-27:
@lfarv I don't think the fix discussed on #895 was ever made, but that is the logic that I am now implementing - I'll put it on a PR on Monday

lfarv, 2025-07-01:
> I checked with MAX IV and it was okay. Here is the lattice that isn't possible to load in case you need to test:
> MAX-IV-lattice.zip

I see the problem, the solution is simple, I’ll look when I’m back home