atcollab/at #721: python windows version installation

ZeusMarti, 2023-12-12:
Hi all,

We have encountered problems installing AT in windows 10 for python versions above 3.9 (for equal or below works). We have tried both "pip install accelerator-toolbox" and locally "pip install .".  You will find an screenshot below. Is it a known issue and we should stick to 3.9?

[image]
[image]

ZeusMarti, 2023-12-12:
By the way we tried it in two different PCs with the same result.

lfarv, 2023-12-12:
Hi @ZeusMarti : the answer is in the error message: you C compiler is too old.

Unlike Matlab, python does not clearly states which are the compatible compilers for a given python version, it just says _"You will still need the C compiler that was used to build Python; typically Microsoft Visual C++."_, which depends on where your binary distribution of python was generated.

I found some compatibility table [here](https://wiki.python.org/moin/WindowsCompilers#Which_Microsoft_Visual_C.2B-.2B-_compiler_to_use_with_a_specific_Python_version_.3F).

Personally, I have no problem with Microsoft Visual Studio Community 2019.

Now this only concerns building from source. Installing from PyPI should not require a C compiler, so I would like to see the error you got using `pip install accelerator-toolbox`, which should install a binary.

ZeusMarti, 2023-12-13:
Thanks @lfarv ,

The Visual C++ version seems to be older than 14, I'm a bit confused, I just asked help to our IT section, will see.

[image]

Indeed finally the remote installation `pip install accelerator-toolbox` worked so we can start doing something.
By the way, this is for a new student we just got. Maybe he could contribute in the future, we are still defining his project (6 months, half time), is there any specific project you think suitable for such profile?

lfarv, 2023-12-13:
> The Visual C++ version seems to be older than 14

What you highlight there is not the Visual C++ compiler itself, but the "Visual C++ Redistributable", which is (I think…) a set of libraries which need to be distributed for running apps compiled with Visual C++.

> Indeed finally the remote installation pip install accelerator-toolbox worked 

Good ! I have no way to experiment with binary distribution on Windows, but apparently you confirm that it works without needing a C++ compiler.

> By the way, this is for a new student we just got. Maybe he could contribute in the future

Excellent. Concerning the future development on AT, I expect a lot from the future meeting(s) of the "board", as advertised by Simon, to identify and prioritise the next evolutions.

swhite2401, 2024-07-30:
This issue seems to be solved, can I close it?

closed 2024-08-26 by ZeusMarti