atcollab/at #724: pyat shift_elem ignores the effect of bending angles

oscarxblanco, 2023-12-14:
Dear all,

I have tried the pyat function `shift_elem` to shift the horizontal position of a bending magnet, and I see that it ignores the bending angle. For example,
```python
bd = at.Dipole('BD',0.8,1e-3)
at.shift_elem(bd,1e-3,0)
print(bd)
```
returns
```bash
Dipole:
        FamName : BD
        Length : 0.8
        PassMethod : BndMPoleSymplectic4Pass
        BendingAngle : 0.001
        EntranceAngle : 0.0
        ExitAngle : 0.0
        NumIntSteps : 10
        MaxOrder : 0
        PolynomA : [0. 0.]
        PolynomB : [0. 0.]
        T1 : [-0.001 -0.    -0.    -0.    -0.    -0.   ]
        T2 : [0.001 0.    0.    0.    0.    0.   ]
        K : 0.0
```
which is wrong. Setting only T1 and T2 to the input shift value ignores the effect of the bending angle. Additionally, the horizontal shift would produce rotation matrices R1 and R2, and all T1,T2,R1 and R2 depend on the point where the shift is applied (typically either the center of the entrance point of the bending).

o

swhite2401, 2023-12-15:
@oscarxblanco shift_elem is a pure translation of the whole element, for a curved magnet the direction perpendicular to the center of the magnet in which case I believe rotations do not apply.
For this point, it seems that only the help shall be improved. Do you agree?

Concerning the effect of the bending angle, could you please develop? It is not clear to me what you mean by this.

Thanks!

oscarxblanco, 2023-12-15:
Hi @swhite2401 ,

now I realized that shift_elem only implements transverse shifts, so, you are right, for transverse shifts rotations do not apply. 
Is there any other function to do longitudinal shifts ?

Wrt to the horizontal shift of a bending, it does produce an effect of T1,T2 and R1,R2 that is not symmetric, and depends on the chosen convention for the shift. For example, if you apply at the center of the bending magnet a shift of let's say 1 mm, the entrance and exit faces of the magnet will be shifted but not exactly 1 mm, and the rotation R1 and R2 are not the identity.

If you instead choose to apply a 1 mm shift to the bending magnet at the entrance, T1 will reflect the shift and R1 will be the identity, but R2 and T2 will not due to the reference system rotation.

oscarxblanco, 2023-12-15:
Here is a numerical example of a 1 mm shift of a 90 degrees bending magnet applied at the center. The angle is quite large and unusual for most applications of pyat but it allows to see values that could be recognized by eye.

First the source in matlab using Simulated Commissioning https://github.com/ThorstenHellert/SC
```matlab
dx = 1e-3;
dy = 0;
dz = 0; % opposite sign to the sixth particle coordinate
atilt = 0.0;
apitch = 0.0;
ayaw = 0.0;
ax =  apitch;
ay =  ayaw;
az =  atilt;
magTheta = 2*pi/4;
magLength = 0.8;
%[T1,T2,R1,R2] = SCgetTransformation(dx,dy,dz,ax,ay,az,magTheta,magLength,'refPoint','entrance');
[T1,T2,R1,R2] = SCgetTransformation(dx,dy,dz,ax,ay,az,magTheta,magLength,'refPoint','center');
T1andT2fromSC=[T1';T2']
R1, R2
```
and here is the output
```bash
T1andT2fromSC =
   1.0e-03 *
   -0.7071         0         0         0         0    0.7071
    0.7071         0         0         0         0    0.7071
R1 =
    1.0000    0.0007         0         0         0         0
         0    1.0000         0         0         0         0
         0         0    1.0000    0.0007         0         0
         0         0         0    1.0000         0         0
         0         0         0         0    1.0000         0
         0         0         0         0         0    1.0000
R2 =
    1.0000    0.0007         0         0         0         0
         0    1.0000         0         0         0         0
         0         0    1.0000    0.0007         0         0
         0         0         0    1.0000         0         0
         0         0         0         0    1.0000         0
         0         0         0         0         0    1.0000
```

swhite2401, 2023-12-15:
Ok now I understand what you mean, would you like that changes are implemented or can we consider this a negligible effect and just document it in the help?

Concerning the longitudinal shift, there is nothing implemented and this can be quite tedious as you change the structure of your ring, for bending magnet in particular this will change the reference particle and lattice layout...
Is this something needed for your studies?

oscarxblanco, 2023-12-15:
The error in T1 and T2 produced when ignoring the bending angle needs to be fixed because it is a wrong implementation.

I would suggest to implement the rotation matrices R1 and R2  because they depend on the longitudinal displacement (either due to a transverse shift on a bending, or due to a direct shift) and beam transverse momenta, $x_{err} = \Delta s\tan(p_x)$, and $y_{err} = \Delta s\tan(p_y)$. I don't see a criteria to say it will be negligible or not in a general case because it will add to the particle transverse coordinates, it might be important for some optics while negligible in others.

The implementation of a dedicated function for a longitudinal shift seems to me relevant because it is part of the errors to defined on a lattice. It might be of lower priority/urgency, though.

At the same time it would be much better to have a single function that calculates T1,T2,R1 and R2 from any combination of displacements on $x,y,s$ and rotations around axes $(ax,ay,as)$. It already exists in Simulated Commissioning through `SCgetTransformation.m` (https://github.com/ThorstenHellert/SC/blob/master/SCgetTransformation.m), so, logically I would think it is out of the scope of pyat, and might be implemented somewhere else. For example in pySC ? (https://github.com/lmalina/pySC)

closed 2025-07-23 by swhite2401