atcollab/at #725: pyAT Distinguish between a sector bending magnet and a rectangular bending magnet

tfs563, 2023-12-20:
Hi all, 

I'm trying to figure out if the Dipole element has a way to identify a bending magnet as either a rectangular magnet or a sector magnet. I looked into the Dipole element within the at.lattice.element documentation website, but I'm not sure which parameters or keyword arguments would need to be included/changed to make the magnet one or the other. My first thoughts were the EntranceAngle and ExitAnfgle arguments could do this, but, to me, it seems like it would change the angle of the incoming beam?

Anyway, some help in figuring out how to do this would be much appreciated!

lfarv, 2023-12-20:
Hello @tfs563,

What defines the dipole as sector or rectangular is the pass method. The "geometrical" description of the magnet as length, bending angle and entrance/exit angles is the same for both. There is a short description in the [documentation of passmethods](https://atcollab.github.io/at/common/ex_passmethods.html):

"
**bending magnet passmethods**
_For bending magnet passmethods, the term “rectangular” refers to the layout of the magnetic field, not to the shape of the magnet or to the angle of the pole faces._

_In a “sector” magnet, the magnetic axis is an arc of a circle following the design trajectory,
In a “rectangular” magnet, the magnetic axis is a straight line (ex.: off-axis quadrupole).
In case of a pure dipole (uniform magnetic field), both are equivalent, so one can use any of the available methods. They give identical results._
"

But I agree it's missing in the documentation of dipole elements.

Note that there are only 2 passmethods for rectangular magnets: [ExactRectangularBendPass](https://atcollab.github.io/at/common/ex_passmethods.html#exactrectangularbendpass) and BndStrMPoleSymplectic4Pass (undocumented)

swhite2401, 2024-07-30:
I move this to discussion