atcollab/at #739: python: ringparam not well identified by lattice_load

lnadolski, 2024-02-19:
Dear All,

the at.save_lattice and at.load_lattice do not behave weel in presence of atringparam

# loading a mfile
When using 
ring = at.load_lattice('ESRF_EBS.m', keep_all=True)

the first element of the ring is identified as a marker instead of atringparam.
atringparam('EBS',6000000000,1);...

In [53]: ring[0]
Out[53]: Marker('EBS', Periodicity=1)

# saving a mfile
When saving the lattice, the code adds each time an extra element (atringparam)

function ring = newlattice()
`ring = {...
    atringparam('EBS', 6000000000.0, 1, 'Particle', struct('name', 'relativistic', 'rest_energy', 0.0, 'charge', -1.0), 'HarmNumber', 992, 'Beam_Current', 0.0, 'Nbunch', 1, 'M_File', '/lustre/work/sources/physmach/nadolski/codes/pyenv/collimation/scripts/pyat/touschekscatteringlosses/test_crotch/ESRF_EBS.m');...
       atmarker('EBS', 'Periodicity', 1);...
`

lfarv, 2024-02-19:
Hi Laurent

The `RingParam` element does not exist in python. So when reading a .mat to .m file, the element is removed and its information is moved to the python `Lattice` object.

In the other way, when saving the lattice in .mat or .m file, a `RingParam` element is added in the 1st position because Matlab needs it.

Here, you are using the `keep_all` keyword. This is intended for tests where you want to have the same number of elements in python and Matlab. To achieve this, the `RingParam` element is replaced by a `Marker` placeholder. But this does not prevent from adding the `RingParam` element when saving back the file.

A solution could be to memorise its origin (RingParam) in the Marker so that it can be deleted when saving the file. For the time being, try avoiding the `keep_all` keyword unless it's really necessary.

lnadolski, 2024-02-19:
Dear Laurent,

Thabks for your reply.
This is not convenient when we go back and forth between files.
In my example, I want to do the following

script1.py reads lattice.m and produces outputlattice.m
script2.py reads outputlattice.m and does other computations

What is the good way to do it in Python if I want to keep the same number of elements?
Should I test each time if RingParam exists and remove it? This is not convenient at all...

lfarv, 2024-02-19:
Agreed, I'll implement the trick I mentioned above. IN the meantime, do you really need the `Marker` placeholder ? Can't you simply assume that the python lattices have one element less (the 1st one) than their Matlab equivalent ?

lnadolski, 2024-02-19:
we use AT in hybrid mode python + matlab
it is a bit annoying to modify the lattice each time we switch from Python to matlab and vice versa.
but I guess there are not so many options.
If the lattice were loaded with the function into matlab, we could add it automatically in that function.

lnadolski, 2024-02-19:
the easiest workaround could be:
- in at.lattice_load: to keep it as a marker to keep the same number of elements both in matlab/python
- in at.lattice_save: to remove it before adding the new ringparam element as it is today.

lfarv, 2024-02-25:
Corrected in #741

closed 2024-02-25 by lfarv