atcollab/at #743: bug is saving lattice

lnadolski, 2024-02-28:
Dear @lfarv 

I suspect one bug.

When saving the lattice,  with at.save_lattice(ring, 'newlattice.m')
Eapertures is written instead of EApertures

ring[0] is  Marker('SOLEIL_II_V2366_SYM1_2023JUL24', EApertures=array([0.0075, 0.005 ]), Periodicity=1, tag='RingParam')

newlattice.m:   
atringparam('SOLEIL_II_V2366_SYM1_2023JUL24', 2750000000.0, 1, 'Particle', struct('name', 'relativistic', 'rest_energy', 0.0, 'charge', -1.0), 'HarmNumber', 416, 'Beam_Current', 0.0, 'Nbunch', 1, 'M_File', 'mylattice.m', 'Eapertures', [0.0075 0.005]);...

Best regards,

Laurent.

lfarv, 2024-02-28:
Corrected in #744. The case of the names of all unexpected attributes (not only EApertures) is now kept unchanged.

Note that the optional `Marker` in ring[0] is completely ignored when saving to .m or .mat files (because of the RingParam tag). Only the `Lattice` attributes are used to generate the `RingParam` (they contain the same information).

lfarv, 2024-02-28:
I add that you should not use EAperture or other attributes in a Matlab RingParam element. It's not a real element, it's there only for the storage of lattice attributes (energy, particles...).

lfarv, 2024-07-20:
Corrected in #744

closed 2024-07-20 by lfarv