atcollab/at #757: fast_ring when having harmonic cavities in the lattice

TeresiaOlsson, 2024-04-23:
Hi,

We have a lattice with both main and harmonic cavities and I have problem with generating the fast_ring. Internally the function get_timelag_fromU0 is called and that fails because the cavities don't have the same frequency.

I have set the cavpts attribute on the lattice to only point towards the main cavities and this works when calling get_timelag_fromU0 directly, but not when it's called inside fast_ring because this attribute is never used inside the function. Perhaps fast_ring also needs a keyword to specify which cavity reference points to use like on find_orbit6?

swhite2401, 2024-04-23:
Hi @TeresiaOlsson ,
I've had similar comment from another user, most likely the same issue.
Could you please send me your script by email? I'll take a look.

Normally, get_timelag_fromU0 should be find a solution with multiple RF systems

TeresiaOlsson, 2024-04-23:
Thank you :)

I have attached my files. The script is super simple but I have some parts at the end of the lattice file which maybe could be the cause.
```
    # Turn cavity and radiation on
    ring.enable_6d()
    
    # Set cavpts to point to main cavities
    ring.cavpts = ring.get_uint32_index('CAV*')
    
    # Set only main cavity phases
    ring.set_cavity_phase(cavpts=ring.cavpts)
```

[fast_ring_test.zip](https://github.com/atcollab/at/files/15077069/fast_ring_test.zip)

swhite2401, 2024-05-06:
Fixed in #758

closed 2024-05-06 by swhite2401