atcollab/at #763: Factor 4 missing in Touschek scattering B2 expression

oscarxblanco, 2024-05-02:
Dear all,

there is a missing factor 4 in function `_get_vals` of at/pyat/at/acceptance/touschek.py lines 
https://github.com/atcollab/at/blob/b016b779cc7bbc377a44165196e1c64a776a1d37/pyat/at/acceptance/touschek.py#L127-L129
```python
    B2sq = bg2i*bg2i*(numpy.diff(bs, axis=1).T**2 +
                      sigh2*sigh2*numpy.prod(bxy2*dt2, axis=1) /
                      numpy.prod(sigb2*sigb2, axis=1))
```
`bg2i` is calculated above as $1/(2\beta^2\gamma^2)$. 
`bg2i` is wrongly put as multiplicative factor of the whole expression.

Those lines should calculate the $B^2_2$ factor in Eq. (34) on Piwinski's article titled 'The Touschek effect in Strong Focusing Storage Rings'. https://arxiv.org/abs/physics/9903034
[image]
where the second term on right hand side is not divided by 4.

oscarxblanco, 2024-05-02:
The second term in $B_2^2$ is a crossed-term (x-y) that depends on the product of energy spread, the dispersions (x and y), and the inverse of the transverse emittances.

It has an impact in lattices with vertical dispersion ($\eta_y\neq0,\eta'_y\neq0$) either due to lattice design, limits on the optics correction or transverse coupling. 

However, not many lattices have vertical dispersion.

I have checked the matlab implementation and it does not have this issue. It also has the answer to an approximation that I would suggest to document also on pyat. In matlab the function `TLT_IntPiw_k`  uses the first term in the the modified Bessel function approximation for large numbers: $I_0 \approx \frac{\exp{x}}{\sqrt{2\pi x}}(1+\frac{1}{8x}(1+ ...))$
it is explicitly written in the comments. In pyat, the same logic is used in the function `int_piwinski` but there is no explanation on why is done like that.

swhite2401, 2024-05-02:
Ok. Thanks for this analysis. I will correct this as suggested then.

swhite2401, 2024-05-06:
Correction proposed in #765

swhite2401, 2024-05-28:
Fixed in #765

closed 2024-05-28 by swhite2401