atcollab/at #767: pyAT Error when calculating damping ring

ustc-cheese, 2024-05-15:
Hello everyone 
I am currently working with storage ring with wigglers. But the radiation_parameter will get "AttributeError: 'Wiggler' object has no attribute 'Energy'". Did I do someting wrong or what?
The wiggler was imported in a mat file reads:
period_length = 0.9/8;
l_und = 0.9;
Bmax = 4.531017900217211;
Energy = 800e6;
WIG2 = atwiggler('WIG2', l_und, period_length, Bmax,Energy,'GWigSymplecticPass');
Thank you all
Best Regards
Weijie

swhite2401, 2024-05-16:
Hello,

This element has issues, see #749, and I would not recommend to use it for the moment.
These issues are being addressed in #759 and will hopefully be released soon. You may get in touch with @ZeusMarti and @joanarenillas for details.

Depending on what you are trying to model, you may alternatively use kick maps or the energy loss element to approximate your wiggler.

joanarenillas, 2024-05-16:
Hello @ustc-cheese,

There is a bug in both `GWigSymplecticPass.c` and its radiative version `GWigSymplecticRadPass.c`, that produces the _AttributeError_ you mention. The current versions in the _master_ branch takes the energy for the wiggler from the element rather than the structure. In the Mex part of the integrators, instead of 

`Energy = atGetDouble(ElemData, "Energy"); check_error();`

there should be

`Energy=atGetOptionalDouble(ElemData,"Energy",Param->energy); check_error();`.

If you just want to get around this issue, you can change this line of code in your local wiggler passmethods. Nevertheless, I would not recommend doing this, since these passmetods have additional bugs which can lead you to incorrect results. As @swhite2401 suggested, I would avoid using the current distribution of `GWigSymplecticPass.c` and `GWigSymplecticRadPass.c`, unless it is strictly necessary.

All these bugs are fixed for both `GWigSymplecticPass.c` and `GWigSymplecticRadPass.c` (in Matlab and Python) in my branch `wigglerintegrator_debugged`. You can base your work in my branch to get around these issues, but I must warn you that this branch includes several other changes (see #759) and has still not been merged to the _master_ branch (we are intending to merge it soon).

Hope this helps,

Joan

ustc-cheese, 2024-05-20:
Thank you all! Hoping for your soon correction!

closed 2024-05-20 by ustc-cheese