atcollab/at #790: Severe bug introduced by #766

swhite2401, 2024-06-25:
PR #766 has inttroduced severe bugs in the file loading system, some file now fail to load on the error:

`AttributeError: property of 'Lattice' object has no setter`

After a quick look the attributes failing are `mat_key` and `mat_file`.  Adding these to the list of excluded attributes seems to fix the problem, but I do not understand why.
The logic has become so complicated that it is presently impossible to understand or debug these functionalities, my feeling is that only @lfarv can do it in a reasonable time.

This is in clearly not acceptable and a complete review and simplification of these functionalities is strongly advised.

Meanwhile @lfarv could you please fix this bug?
I attach a file that was saved with pyAT loads without any problem forversion prior PR #766
[opt_ma.txt](https://github.com/user-attachments/files/15972264/opt_ma.txt)

lfarv, 2024-06-26:
fixed in #791

lfarv, 2024-06-26:
The problem is solved in #791. Release 0.6.1 has been uploaded

closed 2024-06-26 by lfarv