atcollab/at #796: ring.get_lifetime fails if list of reference indexes has only one valid element

simoneliuzzo, 2024-07-12:
despite the help asking for reference points of length > 2,
the function fails also if the length is more then 2 (ex: range(0,4)) but in this range only one element length > 0
may be simply changing the help and giving an error for length < 2 could be simpler for the user.

simoneliuzzo, 2024-07-12:
```
import at
import numpy as np

ring = at.load_lattice('/machfs/liuzzo/FCC/92a_107/Errors_for_paper/FCC_92a_107_cson_scent_QF1B_diag_6D.mat',
                       matkey='r')

mom_acc_refpts = range(0, 4)  # fails, but length more than 2 as specified in the help
# mom_acc_refpts = range(0, 10)  # ok

nturns = 1
emit_v = 10e-12       # 10 pm
cur_per_bunch = 0.001  # 1 mA
zn= 0.67
use_mp = True

Tau, MomAcc, indsMA = ring.get_lifetime(emit_v,
                                        cur_per_bunch,
                                        zn=zn,
                                        nturns=nturns,
                                        refpts=mom_acc_refpts,
                                        verbose=True,
                                        use_mp=use_mp)
```

swhite2401, 2024-08-27:
Solved in #801

closed 2024-08-27 by swhite2401