atcollab/at #800: Compilation error when building from repository

TeresiaOlsson, 2024-07-22:
Hi,

I can no longer build pyAT from the git repository. I'm using a recipe to build it into a container which has worked fine before so something has changed, but the error message is vague so I'm not sure what it is. It looks like it can't compile `atimplib.c` anymore but unclear if the error really is there.

I'm using Ubuntu 22.04 and Python 3.12.4.

Anyone has a guess for what the problem might be?

This is what I get:

```
+ git clone https://github.com/atcollab/at.git
Cloning into 'at'...
remote: Enumerating objects: 33130, done.
remote: Counting objects: 100% (7168/7168), done.
remote: Compressing objects: 100% (1748/1748), done.
remote: Total 33130 (delta 6145), reused 5995 (delta 5395), pack-reused 25962
Receiving objects: 100% (33130/33130), 109.43 MiB | 6.08 MiB/s, done.
Resolving deltas: 100% (24204/24204), done.
+ cd at
+ pip3 install .
Processing /at
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Requirement already satisfied: scipy>=1.4.0 in /usr/local/lib/python3.10/dist-packages (from accelerator-toolbox==0.6.2.dev6+gac016b7f) (1.14.0)
Requirement already satisfied: numpy>=1.23.5 in /usr/local/lib/python3.10/dist-packages (from accelerator-toolbox==0.6.2.dev6+gac016b7f) (2.0.1)
Building wheels for collected packages: accelerator-toolbox
  Building wheel for accelerator-toolbox (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Building wheel for accelerator-toolbox (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [196 lines of output]
      /tmp/pip-build-env-9g_l99ak/overlay/local/lib/python3.10/dist-packages/setuptools/config/expand.py:130: SetuptoolsWarning: File '/at/LICENSE.txt' cannot be found
        return '\n'.join(
      ** Entering set_environ. Config settings: None
      ** MPI: 0
      ** OPENMP: 0
      ** Entering setup.py: ['/usr/local/lib/python3.10/dist-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py', 'bdist_wheel', '--dist-dir', '/tmp/pip-wheel-92kp9e7g/.tmp-294a2uzc']
      ** MPI: 0
      ** OPENMP: 0
      running bdist_wheel
      running build
      running build_py
      creating build
      creating build/lib.linux-x86_64-cpython-310
      creating build/lib.linux-x86_64-cpython-310/at
      copying pyat/at/future.py -> build/lib.linux-x86_64-cpython-310/at
      copying pyat/at/constants.py -> build/lib.linux-x86_64-cpython-310/at
      copying pyat/at/_version.py -> build/lib.linux-x86_64-cpython-310/at
      copying pyat/at/__init__.py -> build/lib.linux-x86_64-cpython-310/at
      creating build/lib.linux-x86_64-cpython-310/machine_data
      copying pyat/machine_data/__init__.py -> build/lib.linux-x86_64-cpython-310/machine_data
      creating build/lib.linux-x86_64-cpython-310/at/matching
      copying pyat/at/matching/matching.py -> build/lib.linux-x86_64-cpython-310/at/matching
      copying pyat/at/matching/globalfit.py -> build/lib.linux-x86_64-cpython-310/at/matching
      copying pyat/at/matching/__init__.py -> build/lib.linux-x86_64-cpython-310/at/matching
      creating build/lib.linux-x86_64-cpython-310/at/plot
      copying pyat/at/plot/specific.py -> build/lib.linux-x86_64-cpython-310/at/plot
      copying pyat/at/plot/synopt.py -> build/lib.linux-x86_64-cpython-310/at/plot
      copying pyat/at/plot/standalone.py -> build/lib.linux-x86_64-cpython-310/at/plot
      copying pyat/at/plot/__init__.py -> build/lib.linux-x86_64-cpython-310/at/plot
      copying pyat/at/plot/generic.py -> build/lib.linux-x86_64-cpython-310/at/plot
      creating build/lib.linux-x86_64-cpython-310/at/physics
      copying pyat/at/physics/ring_parameters.py -> build/lib.linux-x86_64-cpython-310/at/physics
      copying pyat/at/physics/energy_loss.py -> build/lib.linux-x86_64-cpython-310/at/physics
      copying pyat/at/physics/matrix.py -> build/lib.linux-x86_64-cpython-310/at/physics
      copying pyat/at/physics/fastring.py -> build/lib.linux-x86_64-cpython-310/at/physics
      copying pyat/at/physics/nonlinear.py -> build/lib.linux-x86_64-cpython-310/at/physics
[... 142 lines cut ...]
      x86_64-linux-gnu-gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -DPYAT -I/tmp/pip-build-env-9g_l99ak/overlay/local/lib/python3.10/dist-packages/numpy/_core/include -I[] -Iatintegrators -I/usr/include/python3.10 -c atintegrators/WakeFieldPass.c -o build/temp.linux-x86_64-cpython-310/atintegrators/WakeFieldPass.o -std=c99 -Wno-unused-function
      In file included from atintegrators/WakeFieldPass.c:2:
      atintegrators/atimplib.c: In function ‘compute_kicks_longres’:
      atintegrators/atimplib.c:300:27: warning: unused variable ‘loopend’ [-Wunused-variable]
        300 |     int i,ii,ib,loopstart,loopend;
            |                           ^~~~~~~
      atintegrators/atimplib.c:300:17: warning: unused variable ‘loopstart’ [-Wunused-variable]
        300 |     int i,ii,ib,loopstart,loopend;
            |                 ^~~~~~~~~
      atintegrators/atimplib.c: In function ‘compute_kicks_phasor’:
      atintegrators/atimplib.c:388:43: error: ‘I’ undeclared (first use in this function)
        388 |     double complex vbeamc = vbeam[0]*cexp(I*vbeam[1]);
            |                                           ^
      atintegrators/atimplib.c:388:43: note: each undeclared identifier is reported only once for each function it appears in
      atintegrators/atimplib.c:380:14: warning: unused variable ‘is’ [-Wunused-variable]
        380 |     int i,ib,is;
            |              ^~
      error: command '/usr/bin/x86_64-linux-gnu-gcc' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for accelerator-toolbox
Failed to build accelerator-toolbox
ERROR: ERROR: Failed to build installable wheels for some pyproject.toml based projects (accelerator-toolbox)
```

TeresiaOlsson, 2024-07-22:
Looks like it's commit 5028aafee80175c8f207a902b271779c53ee255a where I no longer can compile. Did this fix also require new dependencies to be able to compile?

In my recipe I only do the following and get whatever the default version happen to be.

```
    apt-get update
    
    # Install Python 3 and pip    
    apt-get install -y --no-install-recommends \
       python3 \
       python3-pip \
       
    # Update pip to newer version
    pip3 install --upgrade pip        
       
    # Install general tools Python packages
    pip3 install --progress-bar off wheel
    pip3 install --progress-bar off \
        h5py \
        matplotlib \
        numpy \
        pandas \
        scipy \
        scikit-optimize \
        ipyparallel \
	tables \
	mpl-axes-aligner
      
    # Install ipykernel
    pip3 install --progress-bar off ipykernel
    
    # Install necessary packages to build from git
    apt-get install -y --no-install-recommends \
    	build-essential \
    	python3-dev \
        git   
        
    # Install pyAT from the repository
    git clone https://github.com/atcollab/at.git
    cd at
    pip3 install .  
```

swhite2401, 2024-07-22:
This is strange 'I' should be defined if complex.h is included... @lfarv any idea? Is this a side effect from the numpy 2.0 fix?

swhite2401, 2024-07-22:
by the way it seems you are using python 3.10 and not 3.12

TeresiaOlsson, 2024-07-22:
Ah sorry @swhite2401, I'm actually using both. The recipe builds the container starting from a docker container which apparently has python 3.10 but I also tested it on my computer and there I have 3.12. But same error message for both.

lfarv, 2024-07-22:
@TeresiaOlsson : This looks strange: the exact same conditions (ubuntu 22.04, python 3.12) are [tested and succeed on GitHub](https://github.com/atcollab/at/actions/runs/9956741215/job/27507298850).

In you 1st post, the job seems to stop abruptly, after a compiler warning but without error. I'd like to see a full log. Can you just retry
```
pip3 install -v -e .
```

lfarv, 2024-07-22:
@TeresiaOlsson, @swhite2401 : Sorry, I missed the "I"

TeresiaOlsson, 2024-07-22:
I'm also surprised since it worked on Github. I tried to understand what could be different in my setup but I didn't manage. This is what the full log give me:

```
pip3 install -v -e .
Using pip 24.0 from /home/a3744/anaconda3/envs/test-pyat/lib/python3.12/site-packages/pip (python 3.12)
Obtaining file:///home/a3744/Documents/tests/pyat-test/installation/at
  Running command pip subprocess to install build dependencies
  Ignoring oldest-supported-numpy: markers 'python_version <= "3.8"' don't match your environment
  Collecting setuptools>=64
    Using cached setuptools-71.1.0-py3-none-any.whl.metadata (6.6 kB)
  Collecting setuptools_scm>=7
    Using cached setuptools_scm-8.1.0-py3-none-any.whl.metadata (6.6 kB)
  Collecting numpy>=2.0
    Using cached numpy-2.0.1-cp312-cp312-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (60 kB)
  Collecting wheel
    Using cached wheel-0.43.0-py3-none-any.whl.metadata (2.2 kB)
  Collecting packaging>=20 (from setuptools_scm>=7)
    Using cached packaging-24.1-py3-none-any.whl.metadata (3.2 kB)
  Using cached setuptools-71.1.0-py3-none-any.whl (2.3 MB)
  Using cached setuptools_scm-8.1.0-py3-none-any.whl (43 kB)
  Using cached numpy-2.0.1-cp312-cp312-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (19.2 MB)
  Using cached wheel-0.43.0-py3-none-any.whl (65 kB)
  Using cached packaging-24.1-py3-none-any.whl (53 kB)
  Installing collected packages: wheel, setuptools, packaging, numpy, setuptools_scm
  Successfully installed numpy-2.0.1 packaging-24.1 setuptools-71.1.0 setuptools_scm-8.1.0 wheel-0.43.0
  Installing build dependencies ... done
  Running command Checking if build backend supports build_editable
  Checking if build backend supports build_editable ... done
  Running command Getting requirements to build editable
  /tmp/pip-build-env-dtjeakgy/overlay/lib/python3.12/site-packages/setuptools/config/expand.py:130: SetuptoolsWarning: File '/home/a3744/Documents/tests/pyat-test/installation/at/LICENSE.txt' cannot be found
    return '\n'.join(
  ** Entering get_requires. Config settings: None
  ** MPI: 0
  ** Additional modules: []
  ** Entering setup.py: ['/home/a3744/anaconda3/envs/test-pyat/lib/python3.12/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py', 'egg_info']
  ** MPI: None
  ** OPENMP: None
  running egg_info
  creating pyat/accelerator_toolbox.egg-info
  writing pyat/accelerator_toolbox.egg-info/PKG-INFO
  writing dependency_links to pyat/accelerator_toolbox.egg-info/dependency_links.txt
  writing requirements to pyat/accelerator_toolbox.egg-info/requires.txt
  writing top-level names to pyat/accelerator_toolbox.egg-info/top_level.txt
  writing manifest file 'pyat/accelerator_toolbox.egg-info/SOURCES.txt'
  reading manifest template 'MANIFEST.in'
  adding license file 'LICENSE'
  writing manifest file 'pyat/accelerator_toolbox.egg-info/SOURCES.txt'
  ** Leaving setup.py
  ** Leaving get_requires
  Getting requirements to build editable ... done
  Running command Preparing editable metadata (pyproject.toml)
  /tmp/pip-build-env-dtjeakgy/overlay/lib/python3.12/site-packages/setuptools/config/expand.py:130: SetuptoolsWarning: File '/home/a3744/Documents/tests/pyat-test/installation/at/LICENSE.txt' cannot be found
    return '\n'.join(
  ** Entering setup.py: ['/home/a3744/anaconda3/envs/test-pyat/lib/python3.12/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py', 'dist_info', '--output-dir', '/tmp/pip-modern-metadata-0wvqg36n', '--keep-egg-info']
  ** MPI: None
  ** OPENMP: None
  running dist_info
  creating /tmp/pip-modern-metadata-0wvqg36n/accelerator_toolbox.egg-info
  writing /tmp/pip-modern-metadata-0wvqg36n/accelerator_toolbox.egg-info/PKG-INFO
  writing dependency_links to /tmp/pip-modern-metadata-0wvqg36n/accelerator_toolbox.egg-info/dependency_links.txt
  writing requirements to /tmp/pip-modern-metadata-0wvqg36n/accelerator_toolbox.egg-info/requires.txt
  writing top-level names to /tmp/pip-modern-metadata-0wvqg36n/accelerator_toolbox.egg-info/top_level.txt
  writing manifest file '/tmp/pip-modern-metadata-0wvqg36n/accelerator_toolbox.egg-info/SOURCES.txt'
  reading manifest template 'MANIFEST.in'
  adding license file 'LICENSE'
  writing manifest file '/tmp/pip-modern-metadata-0wvqg36n/accelerator_toolbox.egg-info/SOURCES.txt'
  creating '/tmp/pip-modern-metadata-0wvqg36n/accelerator_toolbox-0.6.2.dev6+gac016b7f.dist-info'
  ** Leaving setup.py
  Preparing editable metadata (pyproject.toml) ... done
Requirement already satisfied: scipy>=1.4.0 in /home/a3744/anaconda3/envs/test-pyat/lib/python3.12/site-packages (from accelerator-toolbox==0.6.2.dev6+gac016b7f) (1.14.0)
[... 366 lines cut ...]
          please submit a reproducible example
          (see https://stackoverflow.com/help/minimal-reproducible-example) to:

              https://github.com/pypa/setuptools/issues

          See https://setuptools.pypa.io/en/latest/userguide/development_mode.html for details.
          ********************************************************************************

  !!
    cmd_obj.run()
  error: command '/usr/bin/gcc' failed with exit code 1
  error: subprocess-exited-with-error
  
  × Building editable for accelerator-toolbox (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> See above for output.
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  full command: /home/a3744/anaconda3/envs/test-pyat/bin/python3.12 /home/a3744/anaconda3/envs/test-pyat/lib/python3.12/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py build_editable /tmp/tmpw6qziftk
  cwd: /home/a3744/Documents/tests/pyat-test/installation/at
  Building editable for accelerator-toolbox (pyproject.toml) ... error
  ERROR: Failed building editable for accelerator-toolbox
Failed to build accelerator-toolbox
ERROR: Could not build wheels for accelerator-toolbox, which is required to install pyproject.toml-based projects
```

swhite2401, 2024-07-22:
Hi both, I get the same error on the GitHub actions now, see PR #801 
Something bad has happened....

lfarv, 2024-07-22:
Change in compiler ?

lfarv, 2024-07-22:
My last successful compilation was [yesterday 14:00](https://github.com/atcollab/at/actions/runs/10027232635)…

swhite2401, 2024-07-22:
I can build at master without any problems on my machine

lfarv, 2024-07-22:
I simply restarted yesterday's job, and today it fails ! So indeed something changed in the compilers on GitHub. And also on you machine, @TeresiaOlsson: very strange indeed.

swhite2401, 2024-07-22:
On grappa with ubuntu 20.04.1 and python 3.11, `pip install -e .` fails

swhite2401, 2024-07-22:
Same machine with python 3.8 is ok

TeresiaOlsson, 2024-07-22:
I discovered the issue when I tried to make a pipeline on gitlab for building a container with pyAT. In my pipeline I start with the image ubuntu:22.04 from DockerHub and then everything else needed is installed inside the recipe using `apt-get`. The only two packages in the recipe I think could potentially affect this is `build-essential` and `python3-dev`.

I have updated my computer recently so if something has changed in these packages it could explain why it failed for me both on gitlab and on my computer.

swhite2401, 2024-07-22:
Sorry, misleading information. What fails is when running `pip install -e .` to install `at` in a conda environment.
Running pip in a python venv seems to be working fine, however I could only test up to python 3.9.

lfarv, 2024-07-22:
However what happens on GitHub (same job working yesterday an no more today) is not related to conda in any way…

swhite2401, 2024-07-22:
Using `_Complex_I` instead of `I` seems to fix it...

TeresiaOlsson, 2024-07-22:
It looks promising. I could also build the `get_lifetime_help` branch without errors.

lfarv, 2024-07-22:
"I" is indeed in the C99 norm. Using "_Complex_I" looks like a temporary workaround… What should we do for the master branch?

TeresiaOlsson, 2024-07-22:
If I did the same changes as @swhite2401 on the master branch I could also compile. So it seems to be the fix. But very strange why the behaviour suddenly has changed.

swhite2401, 2024-07-22:
Well until we understand what is going on I would suggest to implement at least the temporary fix, what do you think?

lfarv, 2024-07-22:
@TeresiaOlsson : for me, it's not a fix, but a workaround a recent bug in gcc (no problem with other compilers, clang or VC). Unless the bug comes from another header (python?) which would undefine "I". But given their order of includes (<complex.h> comes late) apparently it's not the case.

@swhite2401 : I agree, let's do that. I'll be away until Friday, so see with @TeresiaOlsson for reciprocal approvals! On my side, I postpone my two PRs for a few days, we'll see then.

swhite2401, 2024-07-22:
Ok I make a separate PR