atcollab/at #803: VariableMultipole Usage

paul7985, 2024-07-24:
Hi everyone, 

I am currently working with pyAT and I want to implement a kicker magnet which provides a new kick every turn with 

`kick_list = np.cos(2*np.pi*np.arange(N_turns)*(tune_x))`

 and my VariableMultipole usage is as follows : 

`kicker = at.VariableMultipole(family_name='kicker', AmplitudeA=1e-7, FuncA = kick_list, mode=at.ACMode.ARBITRARY)`

or

`kicker = at.VariableMultipole(family_name='kicker', AmplitudeA=1e-7, FrequencyA = (2*np.pi*N_turns*(tune_x)), mode=at.ACMode.SINE)`

```
ring.insert(idx=corrector_idx, elem=kicker)
trajs, _, _ = at.lattice_track(ring, x_0, nturns=N_turns, refpts=refpoints)

plt.plot(trajs[0, 0, :, 0:100])
plt.show()
```
The result is not as expected, it seems like the code only consider the kicker with one of the angles and does not change over turns. I guess there are some things I didn't understand and some things missing in my script. I am very new to this community (first intership in PSI) and I am sorry if the problem is basic to resolve.

Thank you,
Paul

swhite2401, 2024-07-24:
Hello,

Be careful the frequency should be provided in Hz. The kick is then defined as:

` AmplitudeA*sin(2*pi*FrequencyA*t)`

In case you would like to produce a list as a function of turn number

list = ` AmplitudeA*sin(2*pi*FrequencyA*np.arange(Nturns)*t0)`

where `t` is the time and `t0` the revolution duration.

In your case you define the frequency as: `2*np.pi*N_turns*(tune_x)` which could be incorrect for 2 reasons. First because it depends on the number of turns you are tracking and second because it seems that you are confusing tune and frequency (depending on  what you define as `tune_x` but this I cannot guess from your code).

Maybe this is the problem?

paul7985, 2024-07-24:
Hello @swhite2401, 

Thank you for your answer. About the second reason, I indeed want the tune to be the kick angle frequency as I want a resonant excitation along the ring. I already put it to the test thought another code but the problem is that it is really slow (N_turns=~60k). I used a for loop to modifiy the kick angle as follow : 

```
for turn in range(N_turns):
        kick_angle = ppre_kick(turn, tune, amplitude)
        print(kick_angle)
        fk2_activation(ring, kick_angle_x=kick_angle, kick_angle_y=0)
        traj, _, _ = at.lattice_track(ring, r_in=last_coordinates, nturns=1, refpts=refpoints)
        last_coordinates= traj[:,0,-1,:]
        
        trajectories.append(last_coordinates)
```
with ppre_kick func and fk2_activation

```
def ppre_kick(turn, frequency, amplitude):
    return amplitude * np.cos(2 * np.pi * frequency * turn)

def fk2_activation(ring, kick_angle_x, kick_angle_y):
    at.set_value_refpts(ring, refpts=corrector_index, attrname='KickAngle', attrvalues=kick_angle_x, index=0)
    at.set_value_refpts(ring, refpts=corrector_index, attrname='KickAngle', attrvalues=kick_angle_y, index=1)
```

So I wanted to make my code faster and I was told to use 'VariableMultipole'. I think my problem is more about the first reason. I don't understand how to provide a new kick angle for every turn with this command, even though I read the documentation and source code multiple times. I may need a clear explanation of the VariableMultipole command if you have one and a small example could be precious. 

Thank you very much

lcarver, 2024-07-24:
Hello,
Recently I was using:

`varh = at.VariableMultipole('h_noise', AmplitudeB = [1.0], FuncB=kick_table, MaxOrder=0, Periodic=True, mode=ACMode.ARBITRARY)
`

It's important to remember that you can only do 1 order at a time for each VariableMultipole element. This is defined by MaxOrder. So here I just set AmplitudeB to be 1 because then it is basically ignored, I give FuncB my table of Turn by Turn kicks. I know it is a horizontal kick because I give AmplitudeB and FuncB (as opposed to A for vertical) and I know it is a dipole kick because MaxOrder=0.

FuncB should be an array of kicks that will be evaluated at each turn. One potential problem you could have is whether or not you use

```
for i in np.arange(Nturns):
    _ = ring.track(parts, nturns=1)
```

or 
`    _ = ring.track(parts, nturns=1)`

The former may run you into some problems without giving additional flags to track (because in the C part it will think it is in turn 1 every time).

swhite2401, 2024-07-24:
Again, in `ppre_kick `you seem to be using the tune (frequency/frev) so it looks like you are using the wrong quantity.
Anyway here is a code to excite the betatron frequency (to excite the vertical plane you have to use `*A` instead of `*B`)

```
tunes = ring.get_tune()
frev = ring.get_revolution_frequency()
acmpole = at.VariableMultipole('ACMPOLE', AmplitudeB=1.0e-6, FrequencyB=tunes[0]*frev)

xout1, _, _ = ring.track(numpy.zeros(6),nturns=100)

fring.append(acmpole)
xout2, _, _ = ring.track(numpy.zeros(6),nturns=100)

plt.plot(numpy.squeeze(xout1)[0,:], label='No excitation')
plt.plot(numpy.squeeze(xout2)[0,:], label='With excitation')
plt.show()
```

And the figure where you can see the horizontal motion excited as expected:

[image]

paul7985, 2024-07-24:
Thank you for both of your answers. 

I am glad you helped me, it worked. I thought that I would need to give the program an information about the number of turns. 
The answer from @swhite2401 proved me wrong. Thank you for your clear explanations, also about A and B (H/V), I had no idea about that either. I am really grateful. 

I don't want to offend anyone by saying this but the pyat documentation might need some additional information for newcomers. What you wrote in either of your responses is not displayed in the VariableMultipole doc. The small example from @swhite2401 could be of relevance in my opinion. 

Anyways, thank you for your time,
Paul

lcarver, 2024-07-24:
You are welcome, in many places the documentation is lacking and we need to improve it. 
Cheers,
Lee

swhite2401, 2024-07-24:
For sure the documentation would need to be improved we fully agree on that...
I move this issue to discussion, it could serve other people.
Cheers