atcollab/at #805: Inherited fields don't exist unless first explictly set

TeresiaOlsson, 2024-07-29:
Hi,

I'm not sure if this is a bug or if I have the wrong expectation of how things are supposed to work, but if I try to set fields that should be inherited from parents it never works when using `set_value_refpts` unless the field is first explicitly set either when creating the element or afterwards. For example, fields like `KickAngle` and `RApertures`.

Some examples:

```
test_quad = at.Quadrupole('test_quad',0.2,k=2)
ring = at.Lattice([test_quad],name='test',energy=2e9)
ring.set_value_refpts('test_quad', 'KickAngle', 1e-3, index=0)
```
Fails with error `'Quadrupole' object has no attribute 'KickAngle'`

```
test_quad = at.Quadrupole('test_quad',0.2,k=2)
ring = at.Lattice([test_quad],name='test',energy=2e9)
quad = ring['test_quad']
quad[0].KickAngle = [0,0]
ring.set_value_refpts('test_quad', 'KickAngle', 1e-3, index=0)
```
Now it works.

```
test_quad = at.Quadrupole('test_quad',0.2,k=2,KickAngle=[0,0])
ring = at.Lattice([test_quad],name='test',energy=2e9)
ring.set_value_refpts('test_quad', 'KickAngle', 1e-3, index=0)
```
This also works.

Is it supposed to work like this? Especially I find it very confusing when you try to use keyword arguments that the documentation says should exist but they don't unless you have explicitly set them when you created the element.

lfarv, 2024-07-29:
@TeresiaOlsson : this is how it is supposed to work now, even if it is not very convenient…

You can create any **scalar** attribute with `set_value_refpts`, but you cannot create an **array** attribute. you can only modify it.

The problem is that `set_value_refpts` cannot determine the shape of the array from its arguments. For "standard" optional attributes, like `KickAngle` or `RApertures`, there is a pre-defined shape, but no default value, so `set_value_refpts` does not know how to initialise the unspecified items. For use-defined attributes, it's even worse since there is no shape.

We could think of improving that by defining default values. But for now, the solution is to 1st create that attribute (if it's the wrong shape, you get an error), and then modify it if necessary.

TeresiaOlsson, 2024-08-06:
Ah, I see. Thank you :) Now it makes sense why it is working like this. 

I think at some point default values would be useful, but mostly for documentation and pedagogical reasons. At HZB most people are Elegant or MAD-X users who start using pyAT without previous experience of Matlab AT. Since it's so easy to printout a python object I very often see that the first thing that they do when they want to start learning the code is to printout an element and check which attributes they have to work with. And since they don't know which attributes to expect based on what Matlab AT has they sometimes miss attributes they could have used because they are not listed.

closed 2024-08-06 by TeresiaOlsson