atcollab/at #817: Whether there are macroparticles in AT？

pistachio-xin, 2024-08-20:
If AT has the concept of macro particles? I want to track whether a bunch will be lost in the storage ring, but even a 10pC bunch has nearly 100 million electrons, which is too much computation. Is there a macro particle way to track it?

lcarver, 2024-08-20:
Hello.

AT has many PassMethods for tracking in the presence of collective effects. (See https://atcollab.github.io/at/p/howto/Collective.html). When tracking with collective effects you have to use a macroparticle approximation, which applies wake induced kicks from the head to the tail of the bunch based on slicing the ct coordinate.

What kind of effects are you wanting to simulate?

Cheers,
Lee

lcarver, 2024-08-20:
Which machine are you working on? You will need a lumped wake potential for the short range wake or at least some idea of how large it is in order to make an approximation. I recommend that you look in the PyAT examples. You will need to use either 'fast_ring', which converts a full lattice into a few elements, or you can construct a simple lattice using 'simple_ring'. You can refer to the guide above for the documentation. If you want to discuss in more detail how to set up this kind of simulation, you can email me on lee.carver@esrf.fr
Cheers,

closed 2024-08-20 by JBGuan

pistachio-xin, 2024-08-20:
I will try the method you said, thanks again for your detailed answer.