atcollab/at #825: ohmienvelope with radindex length smaller than the ring produces an error

oscarxblanco, 2024-09-04:
Dear all,

it seems to me there is a bug in at matlab `ohmienvelope`.
The following lines use the variable `Wig` to mask `radindex`.
The length of `Wig` is equal to the length of the ring, while, the length of `radindex` could be equal or smaller than the length of the ring.
Matlab does not throw an error, but, does an unexpected operation with the lists of different lengths. Could you confirm it ?
https://github.com/atcollab/at/blob/8ba4eb17ea22c839ab13f9d916b2d548cdc02152/atmat/atphysics/Radiation/ohmienvelope.m#L29-L30

Best regards,
o

lfarv, 2024-09-04:
`radindex` is normally provided by `atenable_6d`:
```
if nargout > 1
    cavitiesIndex=atgetcells(newring,'PassMethod',@(elem,pass) endsWith(pass,'CavityPass'));
    radelemIndex=atgetcells(newring,'PassMethod',@(elem,pass) endsWith(pass,'RadPass'));
end
```
and has the same length as `Wig`, so for me there is no problem.

oscarxblanco, 2024-09-04:
Oh, I see, `radindex` is actually a mask.
I mean not [1,2,3,...] but zeros and ones [0,1,0,1,...]

closed 2024-09-04 by oscarxblanco