atcollab/at #835: PyAT 'PassMethod QuadLinearRadPass: library, module or trackFunction not found'

space-budgie, 2024-09-14:
### Issue

I am trying to analyze a lattice written in Matlab with PyAT. I have generated the lattice in Matlab, exported the accelerator as a .mat file and loaded the lattice in PyAT. I can do 4D tracking without problems, but when I execute `lattice.enable_6d()` and then try to track particles, I get the error:

```RuntimeError: PassMethod QuadLinearRadPass: library, module or trackFunction not found```

### Potential Cause

I believe that `.enable_6d()` changes all pass methods that end with `Pass` to end with `RadPass` instead.

https://github.com/atcollab/at/blob/5e3ca23b90725c24f74fd21a1e75058c9af59b9a/pyat/at/lattice/elements.py#L164-L166

It seems like `QuadLinearPass` is a valid pass method, but `QuadLinearRadPass` is not. 

https://github.com/atcollab/at/blob/5e3ca23b90725c24f74fd21a1e75058c9af59b9a/pyat/at/load/utils.py#L126-L141

I also cannot find any `QuadLinearRadPass` in the `at/integrators` directory.

### Temporary fix

Looking at `at/atintegrators/QuadLinearPass.c`, it appears that `QuadLinearPass` already supports 6D calculations. Hence, I have inserted the code 

```Python
if self.PassMethod == "QuadLinearPass":
    return "QuadLinearPass"
```

above line 164 of https://github.com/atcollab/at/blob/5e3ca23b90725c24f74fd21a1e75058c9af59b9a/pyat/at/lattice/elements.py.

This seems to work. I can enable 6D and track particles without problems. However, this solution is cleary very ad hoc.

lcarver, 2024-09-14:
Hello,

You can do

Ring.enable_6d(quadrupole_pass='QuadLinearPass')

Which should do exactly what you want.

Cheers,

Lee

Links:
------
[1] https://github.com/atcollab/at/issues/835
[2] 
https://github.com/

lfarv, 2024-09-15:
Hello @space-budgie.
`QuadLinearPass` is a simplistic passmethod which does a matrix-based transfer through the element. To model synchrotron radiation, you need a method based on a drift-kick sequence (radiation occurs in kicks). That's why there cannot be a `QuadLinearRadPass` method.
`QuadLinearPass` was the default for quadrupoles a long time ago, but it has now changed for the more accurate `StrMPoleSymplectic4Pass`. Maybe you Matlab lattice was created a long time ago. For now, you have two options:
1. Use @lcarver's way of turning radiation on: you'll neglect synchrotron radiation in quadrupoles, which is good enough when running on-momentum,
2. Better option: convert your quadrupoles to use `StrMPoleSymplectic4Pass`: you get a more accurate tracking, at the cost of a small speed reduction, and you can turn radiation on in quads.

closed 2025-07-23 by swhite2401