atcollab/at #846: pyat linopt6 docstring error in betatron phase range

oscarxblanco, 2024-10-18:
Dear all,
the function `at.linopt6` has a misleading annotation in the documentation stating the value is module $2\pi$.
[image]
In fact the function does not return a value modulo $2\pi$. It could be used to find the total phase advance and therefore the integer and fractional part of the tune, as I show here.
```python
>>> pp = ring.linopt6(refpts=at.All)
>>> pp[2]['mu'][-1]/2/pi
>>> pp[2]['mu'][-1]/2/numpy.pi
array([3.82399794e+01, 1.23562832e+01, 2.37592751e-03])
>>> ring.get_tune()
array([0.23997936, 0.35628316, 0.00237593])
```

lfarv, 2024-10-19:
Strictly speaking, nothing guarantees that you get the total phase advance. It depends on the sampling (number of refpts along the lattice). If the phase advance is more than $2 \pi$ between two refpoints, you may miss it. This is very unlikely if you ask for values at each element (`repfts=at.All`). But if you try your example with `refpts=at.End` (I don't know which lattice you took), you may find a different result.

swhite2401, 2025-07-23:
Solved in #959

closed 2025-07-23 by swhite2401